HiddenRamblings/TagMo #248: Repository Push Access

AbandonedCart, 2021-08-17:
I understand the original author and contributors may no longer wish to maintain the project, which is why I am requesting push access. 

I have merged many of the changes that received a positive response to my own fork, along with updating for both older and newer Android. I have converted most of the application to allow translation and would be more than willing to manage integrating it.

I can easily branch off these changes to avoid replacing any current checkpoints. This is strictly for the benefit of the user, as I feel the current statements about the project still working no longer apply to a large portion of the remaining community.

I also understand that this project may no longer be tracked by anyone with the proper permissions to grant this request. I will continue to submit my changes to the PR for the next 72 hours before converting this issue and the PR into a redirect. I will retain any and all credit to the original contributors and will gladly respect any requests by the original authors to discontinue any or all actions.

Thank you

HiddenRambler, 2021-09-10:
@AbandonedCart It's true i currently lack the time to maintain this project. Would you like commit acccess to this project?

AbandonedCart, 2021-09-10:
That's up to you. I'm not sure the automated signing will port over to this repository without a key being set up, but that isn't significant compared to the progress.

HiddenRambler, 2021-09-10:
I have sent an invite please double check and let me know if it's working. (It's been a while since i've dealt with github permissions)

AbandonedCart, 2021-09-11:
It worked. I am closing down the notices and moving the updated code to a new local branch. Thanks

closed 2021-09-11 by AbandonedCart