HiddenRamblings/TagMo #252: N2 Elite Compatibility

AbandonedCart, 2021-09-13:
Opening this up for discussion of an attempt to bring N2 Elite compatibility to TagMo.

Disclaimer: I do not have the hardware to personally test any code. At best, my contributions are based on documentation, example code from various sources, and reverse engineering the official app.

burndive, 2021-09-13:
I have the hardware, and I would be eager to test any features or builds.

Similar to the official N2Elite app, after scanning an N2Elite, TagMo should be able to display the amiibo assigned to each of the slots.  There should be an interface for selecting which slot is active, as well as assigning amiibo bins to slots.  

Ideally, after scanning an N2Elite, you could reassign as many slots as you wanted, before writing them all in a single scan action.  The official N2Elite app only allowed rewriting one slot at a time (per scan), but I think the API lets you do them all at once.

HiddenRambler, 2021-09-13:
Is this related to the PowerTag or a completely separate product? It should not be too difficult to reverse engineer the app to integrate this feature into TagMo

AbandonedCart, 2021-09-13:
It's similar, but poorly documented. It's also hard to support all the features, but you can already see how it's expected.

HiddenRambler, 2021-09-13:
I just tried a quick test trying to program a powertag i had with the N2Elite app which failed. So looks like they are different. Unfortunately the N2Elite seems to be way too expensive and not worth it. There is much more cheaper alternative called MagicNTag (https://lab401.com/products/ntag-compatible-21x) which is much cheaper, but the last time I checked it had a hardware bug. They promised to fix it soon but still waiting for an update from them.

AbandonedCart, 2021-09-13:
I didn't buy one for the same reason. As much as I like doing local testing, I couldn't justify buying that over another bag of 100 tags for 1/10 the price.

ca1e, 2021-09-16:
it just have some specified ntag commands for slots management. I already contributed these codes but the UX still need be done.

AbandonedCart, 2021-09-16:
The only problem is that the code requires specifying memory blocks. It's not as simple as single tags, so the usage part ended up being a decent size job.

BlackRabbit22, 2021-09-16:
I don't mind doing any tests either as ive got an n2 elite as well.

Catflap62, 2021-09-28:
I have a spare Amiiqo (N2 elite original name) I am willing to send if it will help get compatibility in tagmo.

AbandonedCart, 2021-09-29:
I can't promise anything, so it's really up to you if you want to go giving it away. A lot of the delay in progress recently has been medical stuff, but it should only be a temporary delay.

Catflap62, 2021-10-01:
Hi
Thanks for replying, as the official app is dead and I don't have an iPhone
to use Ally, any hope of getting it working in TagMo is worth it.
If you want to send details I will send it.
Thanks
David

AbandonedCart, 2021-10-01:
@Catflap62 My email is twistedumbrella@gmail.com

Catflap62, 2021-10-08:
Posted, hopefully about a week.

AbandonedCart, 2021-10-16:
Arrived today. Thanks again. Looks like I know what i'll be doing tonight.

Catflap62, 2021-10-16:
Great, thanks for letting me know, hopefully it will help you.

Catflap62, 2021-10-16:
I have the latest MCAS (Most Complete Amiibo Set) from October 7th if you need it.

AbandonedCart, 2021-10-16:
I should be alright there, but thanks.

AbandonedCart, 2021-10-16:
... so minus the time spent banging my head on the wall for why it wasn't working before realizing I hadn't enabled the option I added for it, I have the basic capability to swap banks, set bank count, and read / write individual tags working.

AbandonedCart, 2021-10-16:
It's still pretty basic and very VERY ugly, but it's a start.

AbandonedCart, 2021-10-16:
Was the N2 / Amiiqo able to read every slot with a single scan in the stock app? It seems like an ugly process to scan 200 tags individually. I would assume it did, but the current state of the official app makes it seem like it didn't.

Catflap62, 2021-10-17:
Yes, it did. I use version 2.0.11 now, because it has the image database built in, so at least I can some Amiibo images. You can find it here
https://android-apk.org/com.n2elite.manager/44578303-n2-elite-mastercheaterz/

AbandonedCart, 2021-10-17:
Thanks. That's good to know. I would rather not quite know how to do it right now than find out it wasn't possible lol

Sora1995, 2021-10-17:
Have a N2 as well. Awesome, can't wait for full functionality. Perhaps it's just mine, but it seems setting the bank count is the top one, while setting the bottom is the actual bin, although it seems like some issues have arisen from it. Normally, when you lower the bank count, it doesn't delete the bins from the slots. This one seems to have. Not a big deal as I have backups, but worth noting

AbandonedCart, 2021-10-17:
Unfortunately, that is how the code appears to work. It's the same code as the N2 Elite app itself, so I'm assuming the only reason that appears to not delete them is a cached copy of what was there saved in the app preferences.

Sora1995, 2021-10-17:
Hmm, perhaps I'm just having bad luck with it. Installed the older N2 app (was using the broken one) and they were all still there and accessible. Disregard

Sora1995, 2021-10-17:
Hmm, I keep getting this. I'm assuming it's just a problem with the N2? 
[image]

Sora1995, 2021-10-17:
So, if a slot is selected that doesn't have a . bin saved to it, this shows up
[image]

Some more testing is definitely necessary. One thing (if possible to add) would be a view of your bank. Being able to easily swap it on the fly would be really great for those that do Amiibo tournaments. Pressing the button on the N2 is *supposed* to switch to the next Amiibo (though mine could be busted and it is in alpha). An option to input how many slots we want available as well as a slider would be good as well (they can be finicky if you have a weird number), and I think the app remembering the last option you picked for what bank would be good so that it doesn't always restart to 0. Apologies for so many requests, I'm just thinking of what would make this app easier to use with the N2 >.>

ca1e, 2021-10-17:
- set bank count
- show all slots info
- random serial number

ca1e, 2021-10-17:
> Was the N2 / Amiiqo able to read every slot with a single scan in the stock app? It seems like an ugly process to scan 200 tags individually. I would assume it did, but the current state of the official app makes it seem like it didn't.

the official n2 app is just read each slot in loop in one scan, and it just read 8 byte for the serial id info (in the page 21&22)

Sora1995, 2021-10-17:
> The hardware arrived last night, so the focus was getting the basic functionality running. Getting it to where this is a real replacement for the stock app requires a whole new UI. That takes a bit longer than a couple buttons and some one line commands.

Not a problem at all. Just food for thought!

Catflap62, 2021-10-17:
Should not have bank 0 should start with  bank 1.

AbandonedCart, 2021-10-17:
Alright, I made a list of the features that need to get implemented at the top of the page. That will act as somewhat of a roadmap on what is being done.

AbandonedCart, 2021-10-19:
I think we're going to start off with a somewhat basic list of what's on there that allows read / write / edit of each item and grow it from there. Writing the whole thing will come after that. The idea will be to get features out as quickly as possible.

Either before or after write, the firmware update will (at least attempt to) be added. In theory, the stock app should be fully replaced, since they don't seem to be updating that anymore.

Ericzion1999, 2021-10-19:
How i write on It?

Ericzion1999, 2021-10-19:
> @Ericzion1999 As it says above, writing is coming next. For now, click a slot and use the single item menu.

Thanks you , and sorry 😅🙈

burndive, 2021-10-19:
The display aspect is beautiful.  I would suggest that you add slot numbers to that display, along with whatever management UI is needed for manipulating each slot as features are added.

Also, the only way I could trigger it to recognize that it was scanning an N2 Elite was to uncheck, and then check "Enable N2 Elite Support" in settings before scanning.  Previous to that it didn't recognize the device as a tag at all.

The bank count slider is a bit difficult to get to an exact number.  In the future maybe change that to an editable number field with -/+ buttons.

AbandonedCart, 2021-10-19:
By the way, that slider is trash. It is tedious, finicky, and generally the worst "widget" I've included in any app. That said, no such thing as a NumberPickerPreference really exists and writing one really needs to be done after I know anything broken is the picker and not the actual feature. Hence... crummy, stupid, frustrating slider.

AbandonedCart, 2021-10-20:
Pushing firmware flashing through ASAP (after local testing) because that is the one thing that makes the N2 elite app fix when anything else breaks and you end up with that "not a tag" error that fixes itself after a scan with that app. Once that goes live, features will start coming out of beta. The big roadblock so far has been relying on the old app as a restore.