HiddenRamblings/TagMo #253: New version and PowerTag.

KikiNeosquall, 2021-09-27:
Hello.
The new version don't write on my PowerTag, but the old (2.7.0) yes.
Any ideas ?
PowerTag support is enabled.
Thank you ! 😃

HiddenRambler, 2021-09-29:
I will attempt to test this in the weekend. I need to find my old powertags.

AbandonedCart, 2021-10-18:
Alright. Thanks. I'll look into it.

XiNaru, 2021-10-19:
which version must i download=? the experimental AND the normal release are bothe 10 hours ago :)

AbandonedCart, 2021-10-19:
What is the phone and Android version this is running on?

AbandonedCart, 2021-10-21:
sdcard is the name of the folder with or without a physical card. Neither of mine have one, either.

AbandonedCart, 2021-10-21:
[image]

XiNaru, 2021-10-21:
THERE IS this one or not with me. you say the sd folder is created after the logcat file has been created, but it doesn't.

I have already created the file several times:
[image]
[image]

AbandonedCart, 2021-10-21:
The folder isn't actually empty, but that browser blocks viewing it. Some did that. Looks like I will have to add a way to get to it into TagMo

AbandonedCart, 2021-10-21:
Some file browsers treat it as app-specific storage and don't request the permission to view it.

XiNaru, 2021-10-21:
understand. i can't find a way to get around that on the internet either. despite developer options.

XiNaru, 2021-10-21:
> Got it. Thanks.

noi problem. i hope you can fix this problem
[image]
: