HiddenRamblings/TagMo #278: Randomize serial (WIP)

AbandonedCart, 2021-10-23:


iainsmacleod, 2023-07-01:
Can someone explain why this is closed. I'm still seeing the once per day error, indicating that the serial is not being randomized. Just looking to understand. Thanks

AbandonedCart, 2023-07-01:
The feature has been available for two years.

iainsmacleod, 2023-07-01:
That being said, why do I see the error in TOTK stating I can only use the amiibo once a day? Thanks for the prompt response fwiw

AbandonedCart, 2023-07-02:
Could be any of 100 reasons. Could be a Power Tag, an old build, trying to update a tag that's already written. Seems like a good place to start would be to install the newest version and follow the instructions. From there, you can submit a bug report if it doesn't work. I have a feeling it won't come to that, though.