HiddenRamblings/TagMo #304: No Amiibo Fount

reversedtardis, 2021-10-30:
I don't use Github so I apologize if this isn't where this needs to go. Upon importing the keys, the app says they are found, but not that they are ok (guides I have found online have images with it saying the keys are ok, so I don't know if that is an older version thing or not). There is no option to uncheck Enable Amiibo File Browser at all. I have downloaded the keys twice from different sources, selecting different ones after it did not work the first time, but there is still no option. I have attempted to reset amiibo info and then sync them, to no avail. Is it something wrong with my phone? It is a Google Pixel 4a 5g, which to my knowledge, has NFC capabilities. Thanks!
[image]
[image]

AbandonedCart, 2021-10-30:
It says the keys are found. Where do you have the Amiibo stored and do you have both an /sdcard and physical sdcard directory?

reversedtardis, 2021-10-30:
I have the amiibo stored in my phone's download folder (I just downloaded them and found them in my phones download folder via the import keys section), I do not believe I have an SD card, as I dont think my phone has a slot for one.

AbandonedCart, 2021-10-30:
Alright. If you have a lot of files or a lot of internal storage, the initial load may take some time. Currently there is a bit of a bug with the file browser (that bottom bar) not working on the first load. Once you exit and return to the app, you can use that to pick the folder where the Amiibo are located to make it load faster going forward.

reversedtardis, 2021-10-30:
Aha! Thanks for that, I think I got them to work by choosing where they are located (won't know for certain until the tags arrive). It may have been just that, them not loading very fast. Trying to make a backup of one of my cards (san rio) didn't work, I'm guessing because the files don't have them listed. Thanks for your help!

Catman-232, 2021-11-02:
I'm having this same problem now after updating to the latest version on the releases page (1688e14), showed all the bin files previously but now doesn't. I've got an SD card but had the files in internal storage because they wouldn't show up otherwise. It seems as if the Emulated Storage Mode checkbox won't stay off if you click it off, opening the settings shows it checked again. Right now I have the files in both locations and it is trying to get them from internal, but using the directory pull up to focus it on where they are still comes up with nothing...

AbandonedCart, 2021-11-02:
Emulated Storage Mode enables itself when no files are found om physical storage. 
Pick one location for the files and go from there. 
2.7.0 is two years old.

Catman-232, 2021-11-02:
Alright well whatever, I started fresh with the previously mentioned one. It just doesn't find them. "No Amiibo Found". The folders are all there like before when it worked.

Now on the new version it just crashes. I install, open the app, allow it to view files, get the prompt about the keys, then the AmiiboAPI prompt, sync to that, then I select the keys. I see a blank amiibo list, with the file directory label also blank, if I refresh it crashes and keeps crashing on startup unless I reinstall.

AbandonedCart, 2021-11-02:
Can you send a log? It's in the three dot menu on the main screen.

Catman-232, 2021-11-02:
Had to make it before causing the crash loop, so hopefully it still has some useful information in it 
[tagmo_logcat[1].txt](https://github.com/HiddenRamblings/TagMo/files/7462055/tagmo_logcat.1.txt)

AbandonedCart, 2021-11-02:
The log was pretty much empty, but I made a few changes to prevent startup crashing

Catman-232, 2021-11-02:
Works as expected now, wish I could provide more useful feedback as to why it now works.

AbandonedCart, 2021-11-02:
I have a feeling I know, but I appreciate it.

closed 2021-11-02 by AbandonedCart