HiddenRamblings/TagMo #305: Allow the ability to pick amiibos from the main menu (N2 elite)

BlackRabbit22, 2021-10-30:
Please allow for the ability to pick amiibos main amiibo selection screen instead of having to:
Scan the n2 via the browser -> 
clicking on the desired bank -> 
viewing the current amiibo bank -> 
clicking on the three dotted menu -> selecting replace amiibo.

AbandonedCart, 2021-10-30:
The best you are going to get right now would be

Scan the n2 via the browser ->
clicking on the desired bank ->
viewing the current amiibo ->
selecting replace

As much as listing it out is meant to make it sound tedious, it handles an equally long list of common issues.

Selecting a bank should only apply to N2.
You can't write beyond the actual bank count.
You could potentially replace an Amiibo you didn't want replaced.
The Amiibo you selected isn't one you have a file to write.
You made changes in another app that aren't logged.

And you can replace multiple banks between scans of the unit. It's only there to make sure this app knows the state.

BlackRabbit22, 2021-10-30:
@AbandonedCart 

> Scan the n2 via the browser -> clicking on the desired bank -> viewing the current amiibo -> selecting replace

This sounds much less tedious and would be a good solution. 

> Selecting a bank should only apply to N2. You can't write beyond the actual bank count. You could potentially replace an Amiibo you didn't want replaced. The Amiibo you selected could be one you don't have a file to write. You may make changes in another app that aren't logged.

Thanks for taking the time to explain the reason behind the decision to have it designed that way.

AbandonedCart, 2021-10-30:
I have looked into a "cache choices and write after a scan" process (similar to how editing data for an N2 bank makes you scan it first) and still ended up with a handful of items that can't be verified and a lot of room to make someone go through all that to find out it wouldn't work. There is a lot of stuff that simply can't be known until the scan starts.

The part that makes everything difficult is that the N2 came with an app and a lot of people using it in TagMo started out with that app. It's not like the Power Tags that only have one Amiibo and it's hard to keep a reliable cache of what is on the unit knowing other apps may change it. I would have liked to reference items in the browser list to a bank, but there was no way of keeping it accurate enough to do anything useful.

AbandonedCart, 2021-11-16:
The more we're able to merge and simplify the code, the more realistic it would be to have some functionality like this, if not this exact design.

AbandonedCart, 2021-11-28:
The current process is -
scanning the device (required to know what's on it)
Selecting a bank (required to know where to write)
Selecting write (required to know that's what's being done)
Selecting at item.

It's not much different, but happens in two layouts without jumping around.

The only improvement to really make would be the ability to select an item and offer a bank selection if you're about to write and the device is an N2. This would require two scans on the write, though, since the first would have to halt to prevent corruption. I'll add the necessary UI components for that.

closed 2021-11-28 by AbandonedCart