HiddenRamblings/TagMo #309: N2 Elite documentation - Complete?

AbandonedCart, 2021-10-31:
https://github.com/HiddenRamblings/TagMo/wiki/N2-Elite-(or-Amiiqo)

This is the current documentation. It includes
- Setup
- Selecting an active bank
- Backing up Amiibo from bank
- Writing Amiibo to single bank
- Wiping Amiibo from bank
- Setting the bank count
- Writing Amiibo to all banks

Is there anything else commonly done with this thing (besides lock / unlock) that needs to be added?

choss, 2021-11-01:
Couple of comments: 
- the set the bank-count to 200 workaround could be mentioned, in case there are still issues with that (on fresh re-install of tagmo)
- The bank selection has issues, if you have no amiibos in your tagmo library: should be mentioned, too

Other than that: I like the documentation! Fits all normal use-cases perfectly and works quite nicely, to be honest. 
The lock/unlock nobody should be using anyways at the moment (unlocked is fine for all needs)

AbandonedCart, 2021-11-01:
I'm not sure I know what workaround thjat is.

choss, 2021-11-01:
I mean in regards to https://github.com/HiddenRamblings/TagMo/issues/300

AbandonedCart, 2021-11-01:
There is no workaround on the user side of it. It's all done behind the scenes.

AbandonedCart, 2021-11-01:
I'm also hoping to take care of any issues without any files present. That's really not something that should be a problem. I'd rather fix it than have it become documentation.

choss, 2021-11-01:
From what I had when no amiibo were present, was that clicking on a bank just threw up a black screen. I don't know how representative that is though. 

Once I had any amiibo present, clicking worked. This was confusing for me as I tried to switch banks, only :) 

Also I tried with the latest (stable) release as of yesterday and the initial scan still fails, when reinstalling the app. Again also might be an issue on my end.

AbandonedCart, 2021-11-01:
I'll keep testing. It might be something that got fixed in the cleanup and hadn't been pushed yet. I was testing with unreleased code.

burndive, 2021-11-01:
Instead of "Write to all banks", would it be possible to have "Write to a range of banks" where you can set the starting and ending bank number?  I would want to use this feature to add a series of the same amiibo at the end of the banks, but not to overwrite my existing N2 amiibo at the beginning.  Also, I don't think this feature is of much use unless each bank has a different random serial number, since if they are the same, the console would simply perceive sequential the scans of different banks as being the same physical amiibo.

burndive, 2021-11-01:
Another feature not documented in the Wiki: 
View the data in a bank (such as Nickname, date, and game data) - This is important for being able to manage the slots.
Related: Make Edits to the bank data (such as setting the Nickname and randomizing the serial number) and save it back to the bank

choss, 2021-11-01:
> I'll keep testing. It might be something that got fixed in the cleanup and hadn't been pushed yet. I was testing with unreleased code.

Thanks for taking my (not that good) bug reports into account! I can only say: Very good work on the N2-Stuff :) I love it!

AbandonedCart, 2021-11-01:
It made perfect sense, but it also helps that I knew that I was favoring file references there and needed to change it.

AbandonedCart, 2021-11-02:
Looks like this is complete enough for the current set of features.

closed 2021-11-02 by AbandonedCart