HiddenRamblings/TagMo #310: Cannot Click on files when browsing from TagMo app.

codymitch16, 2021-11-01:
when I click the import keys button, it takes me to my file locations as it should, and I am able to see the files I need, but they are grayed out in all file locations and I can't tap/click them at all!
[image]
[image]

Ericzion1999, 2021-11-01:
Hy , where did you Download the Keys?

AbandonedCart, 2021-11-01:
See if https://github.com/HiddenRamblings/TagMo/commit/cf48b397005f1ba51030138a77cc6ed996c39eeb fixes the issue. If not, we can remove the mimetype helper altogether. It's supposed to help locate the files, not prevent picking them.

AbandonedCart, 2021-11-02:
Closing this unless a new report comes in about it.

greed77, 2021-11-06:
Confirming issue is still in  80da530. Android 12, Pixel 6.
[image]
Unable to select keys from local folder.

greed77, 2021-11-06:
Another interesting thing that I discovered when I was testing, I think I'm able to delete the file from my download folder and then load into tagmo from the trash.

Ericzion1999, 2021-11-06:
Try the Keys from here: [locked_secret](https://drive.google.com/file/d/1j6y1xuccbf3DnNaTs5pw_bYDl6kZvZBD/view?usp=drivesdk)

And 
[Unfixed_info](https://drive.google.com/file/d/1HJ3TiMyW1TBjApjH-b74EgTP906bVpW0/view?usp=drivesdk)

AbandonedCart, 2021-11-06:
Seems the mimetype detection may be broken with certain files. I'll have to test with the "approved locations" further (Downloads, Ringtones, etc).

AbandonedCart, 2021-11-06:
See if https://github.com/HiddenRamblings/TagMo/commit/ec7cac4e54e43333aa2d4ad0804067a219c0daf0 fixes it. If not, I can dump the entire mimeType check. It's supposed to make things faster, but not working is too fast.

greed77, 2021-11-06:
After a little more testing, it looks like it's only when it's in the root of the Download folder. Moving into a subfolder allows me to import the keys just fine.

AbandonedCart, 2021-11-06:
Looks like that build will probably make things worse, not better. I have a feeling it has to do with "Downloads" having special access.

greed77, 2021-11-06:
Probably, permissions are always annoying XD

AbandonedCart, 2021-11-06:
https://github.com/HiddenRamblings/TagMo/commit/d0ee00b5245829c03bf2c1579737606be3679226 should help, though.

codymitch16, 2021-11-07:
thanks for all the help sorry I never replied but I got it to work! thanks!

closed 2021-11-07 by AbandonedCart