HiddenRamblings/TagMo #313: N2 Elite: 3-dot menu is small and hard to press

burndive, 2021-11-01:
When viewing an N2 Elite Bank, the 3-dot menu is small.
When attempting to tap it, half the time I either tap outside and the view closes or tap inside and the Replace menu opens.
[image]

For a non-N2 amiibo, the similar menu is ready to tap:
[image]

AbandonedCart, 2021-11-01:
Padding added in https://github.com/HiddenRamblings/TagMo/commit/93c991c5379109cfc736c072703fa6bf18e1946b

burndive, 2021-11-01:
Tested with 93c991c and 7155225, and it looks and behaves the same as before.

AbandonedCart, 2021-11-02:
Buttons are now centered across the views and N2 view is considerably larger.

closed 2021-11-02 by AbandonedCart

burndive, 2021-11-02:
The menu is entirely gone now. Is that expected?
The only options are: Activate, Scan, Write, Replace

No Wipe, Edit, or Hex

[image]

burndive, 2021-11-02:
With 1ddf118 the menu is back, but it's still difficult to target with a finger per the original issue (although erring to the right no longer closes the amiibo interface).

Also: it looks like there is an off-by-one error displaying the Bank number: "Bank 14" in the screenshot should be "Bank 15" instead. 

[image]