HiddenRamblings/TagMo #314: "Emulated storage mode" setting cannot be disabled

ArutanSipdrae, 2021-11-02:
As mentionned in the title, if your disable it, it reenable itself once you leave the setting.

AbandonedCart, 2021-11-02:
Do you have a physical sdcard?

AbandonedCart, 2021-11-02:
Are your files being stored there?

ArutanSipdrae, 2021-11-02:
Yes

AbandonedCart, 2021-11-02:
Alright. I'll look into it.

AbandonedCart, 2021-11-02:
https://github.com/HiddenRamblings/TagMo/commit/943a7a45d5b65a5003ec05a7a162638e85996104 should help with the balance between the app finding files and those that can't find settings.

ArutanSipdrae, 2021-11-02:
I installed the new release, I could uncheck the setting but it still displayed "No amiibo found" so I deleted all the data, reopened the app, gave permission and reset the keys, but now it keeps saying "Keys not loaded" despite both being found in the settings

AbandonedCart, 2021-11-02:
Can you grab a log? It's the logcat option in the extended menu on the main page.

ArutanSipdrae, 2021-11-02:
So uh I generated the log and it unblocked the app. Now I have all my bin correctly found, but here's the log still

[tagmo_logcat[1].txt](https://github.com/HiddenRamblings/TagMo/files/7463745/tagmo_logcat.1.txt)

AbandonedCart, 2021-11-02:
Looks like it was a permissions issue. Generating the log must have forced it through.

ArutanSipdrae, 2021-11-02:
There's a prompt missing then because when I deleted the data I said yes to all permission prompt (there was only 1)

AbandonedCart, 2021-11-02:
That's the only prompt.

ArutanSipdrae, 2021-11-02:
Oh well, that's gonna be fun to debug ^^"

AbandonedCart, 2021-11-02:
Beats the posts on reddit.

AbandonedCart, 2021-11-02:
Figured it all out. It was less a permission and more a context issue.

closed 2021-11-02 by AbandonedCart