HiddenRamblings/TagMo #316: Blank or garbled data in Edit screen for all N2 banks

burndive, 2021-11-03:
I scanned my SSB Zelda amiibo, and it correctly showed the Nickname and Mii data, as well as Smash Bros Wii U data.
When I Backup the amiibo, the same data is saved to a bin file.
When I write that data to an N2 bank, it shows up correctly, and when the bank is active, my Switch recognizes the amiibo with the same data on it.

When I select Backup I'm TagMo, the data retrieved from the Bank is the same as the data I saved to it.

However, when I select Edit Properties for that Bank data, and scan the N2 to read the Bank, the amiibo is displayed as blank, as if it has never been registered to a console and as of it has no game data stored to it.

The bank gets scanned when Edit Properties is selected, so clearly the app has access to the full data stored in the bank.

This same phenomenon is true for ALL of the amiibo I have stored in my N2 Elite: some of them show blank data, some of them show random-looking characters and numerical values (dates, write count, etc.). It looks like the data is not being decrypted.

closed 2021-11-03 by AbandonedCart

AbandonedCart, 2021-11-03:
https://github.com/HiddenRamblings/TagMo/wiki/Submitting-Issues
> Duplicates of open and previous (user) issues will be closed without warning.

#276
#282
#287
#294
#302
#312

burndive, 2021-11-03:
Just to be clear, these are my own amiibo, that I have scanned and populated to banks in the N2 Elite. There is ZERO PIRACY involved in reproducing this issue.

burndive, 2021-11-03:
Here's my real actual amiibo:
[image]

And the data in a bin file when it's scanned:
[image]

And the data shown when it's saved to an N2 Bank
[image]

burndive, 2021-11-03:
#276 is a feature request to display additional fields in the bin file browser, that you have ultimately declined to implement
#282 is about the app using bin file data for display when it should be displaying N2 data (similar to this issue in that N2 data isn't displayed, but distinct in that bin file data was being substituted--this has thankfully been remedied)
#287 is about different (but coherent) amiibo data being loaded when Edit Properties is selected (similar to #282)
#294 is still about displaying different data (a duplicate of #287)
#302 is about a failed backup operation (not remotely related)
#312 is actually about this same issue, but you immediately closed it thinking it referencing #276 (which is not a duplicate)

Again: this issue is NOT about
- Displaying Nickname and Mii data in the bin browser (rejected--fine whatever, I can deal with it)
- The fact that some amiibo seem to have extra characters at the end of their nicknames (see "Zelda TZ" above)--this seems to simply be an artifact of how the Wii U used to write to amiibo, and TagMo is correct to display the data as it is encoded.
- TagMo behavior in which bin file data was displayed instead of N2 bank data

This issue IS SPECIFICALLY about:
- TagMo not correctly displaying the data that is stored to an N2 bank when "Edit Properties" is displayed.
This issue manifests in one of two ways, depending on the original bin data:
1. The amiibo is displayed as if it has no data
2. The amiibo is displayed as if it has random data (dates are far into the future or past, write count is a high number, nickname and Mii name display as random characters--often from the Chinese character set, since there are so many of them).

burndive, 2021-11-03:
The second problem mentioned in #317 is this issue