HiddenRamblings/TagMo #317: Bank number and edit properties

Catflap62, 2021-11-03:
Scanning N2 Elite shows correct bank numbers, but selecting a bank shows bank number one lower than it should be.
[image]
[image]

Selecting edit properties on a bank, scans the bank again but does not get correct properties. The Midna and Wolf Link amiibo I have on bank 1 has 40 hearts, but if I edit properties and scan bank again I get this
[image]

But if I turn off N2 Elite in settings so Tagmo only reads the active bank when I edit properties I get the correct properties
[image]

AbandonedCart, 2021-11-03:
It would appear that Mii, nickname, and write count are the only 3 fields being "decoded" and not simply read as raw bytes. I haven't tracked the cause, but it may be in how the bytes are collected during the read, since the difference between turning the elite support on and off changes that.

AbandonedCart, 2021-11-03:
I have to tear apart the entire read process and try to recreate the tag read in a way the N2 will accept. This may take a while.

Catflap62, 2021-11-03:
Do we need the folder icon where it shows N2 Elite ID, hopefully without it the whole ID would fit, I don't see the point of Scan as when you select Edit Properties or View Hex Code it will ask to Scan anyway, maybe Delete instead and all moved slightly to left so 3 dots is slightly easier to select. Activate Delete Write Replace ⋮ 
Sorry to be picky just trying to get it looking perfect, as I said before I am amazed at what you have achieved so quickly, many thanks.

AbandonedCart, 2021-11-03:
I can try moving it left more, but I may have to use some scaling tricks. It seems the average screen here is much smaller than mine.

[image]

Catflap62, 2021-11-03:
Yes, you can see the difference in my screenshots, my phone is a Samsung Galaxy A71.

Catflap62, 2021-11-03:
Just been messing with font sizes on the phone and I can get it too fit now by going down one, if I go down three it looks close to yours but my eyes don't like that. So not a problem now.
[image]

AbandonedCart, 2021-11-03:
You shouldn't have to change the font size of the whole phone over it. I'll try to get it to scale down a little smaller than the rest of the app. We can call it an "excess of information" issues

AbandonedCart, 2021-11-04:
See how https://github.com/HiddenRamblings/TagMo/commit/35ec156266e47f9bb92ec3f4cb635bb5a5e6ea38 looks. Android emulator is broken. Getting ready to downgrade soon, since they know and haven't fixed it.

burndive, 2021-11-04:
Things are visible, but the text is a bit too tiny. Maybe dial it back a notch.

[image]

Catflap62, 2021-11-04:
That definitely definitely works and makes it look like yours, but it is just a bit too small for my phone.
[image]

AbandonedCart, 2021-11-04:
Yeah. That's smaller than I was hoping. Let me try to get an emulator that works.

AbandonedCart, 2021-11-04:
I'm not sure if the changes will fix it or cause the entire thing to crash and burn. 

It's some relatively dated code from a translation app I wrote a few years back. It wasn't physically possible to create the UI unless everyone used the dpi trick in their build.prop and not everyone wants that universally. It's meant to bring smaller screens into scale with a dpi mod, but leave phones using one or with screens larger than a Note 7 as is.

Catflap62, 2021-11-04:
That made it even smaller unfortunately.
[image]

AbandonedCart, 2021-11-04:
Oh ok. Looks like I went the wrong direction. That's an easy fix

AbandonedCart, 2021-11-04:
Might make more sense if I make a UI for the value and let it be configurable.

Catflap62, 2021-11-04:
Too far the other way.
[image]

Catflap62, 2021-11-04:
I have been going back through old builds because I was sure I remembered one that did read the properties correctly on an N2 Elite.
Build 1a2b0a5 does it.

AbandonedCart, 2021-11-04:
I wonder if it's not detecting the elite properly in the handoff then.

Catflap62, 2021-11-04:
I just remember it wasn't working then did then wasn't again, so is there some code different in that build?

Catflap62, 2021-11-04:
93e6505 works then the one I have after doesn't e6aa946.
When you select edit properties in e6aa946 it goes to the scan tag screen itself, the builds that work tell you 'Amiibo data refresh required!'

AbandonedCart, 2021-11-04:
That lines up with it being a merge issue.

As for scale, it's not in settings. The min is 10 and max is 30. What you said was even smaller was 15 and too much the other way was 30. The original was 20. I set a default of 25.

Catflap62, 2021-11-04:
Slider seems to do nothing to the actual view.

AbandonedCart, 2021-11-04:
I remember the last slider being a pain, too. They really don't like to save values and display them. Always one or the other.

Catflap62, 2021-11-04:
Slider works but is all over the place and is messing up all views.

AbandonedCart, 2021-11-04:
I think I have the scale all worked out and I put in a workaround for the scanning. The N2 Elite app never legitimately scans Amiibo data, except as a blind read for a backup.

burndive, 2021-11-04:
Scanning works! Edit Properties is displayed correctly.  

Saving the edited data and then performing a Write operation resulted in the bank data being wiped, unfortunately.

Scaling is still smaller than I like set at 10 (slid all the way to the left).

If I set it larger, it gets super tiny, but then if I set it back to 10, it stays that small until I clear the app data entirely.

burndive, 2021-11-04:
Here's the default size, without any adjustment after clearing app data and launching for the first time.
[image]

AbandonedCart, 2021-11-04:
@Catflap62 As is, only the N2 viewer is going to be scaled and only because it has a lot more / lengthier buttons. The scale can be expanded to basically anything, so let me know if anything else looks cramped, but doesn't have items you can afford to lose. I finally got the emulator up, so I can see things.

The only way the serial fits without scale is to remove the icon and also kill the arrows. Putting the active bank there seems like it could cause confusion and some obligatory "slide me up" text could cause hiccups, since it would be displayed and removed depending on slide state.

Catflap62, 2021-11-04:
When I tick Scale N2 Browser View the slider is on 28, when I move slider it changes number, I go back to try but view size is the same, check settings and it shows 28 again.

Catflap62, 2021-11-04:
Edit properties works now.
Format should be Delete and still don't see the point of Scan Bank option in 3 dot menu.

AbandonedCart, 2021-11-04:
You're formatting the babk, not deleting it. Delete sounds like bank will be gone entirely.

Scan is there because it is once and done per instance. For someone that doesn't want top hit edit and find out it scans, they can scan first and then edit simply opens.

AbandonedCart, 2021-11-04:
We're going to close this one for now and a new one can be started for scale and menu stuff, if necessary. I think we've addressed the original issues, though. I'll be pushing through some cleanup shortly (~20 minutes).

closed 2021-11-04 by AbandonedCart