HiddenRamblings/TagMo #319: When changing banks on N2 Elite tagmo does not scan anymore until reopened

choss, 2021-11-04:
Steps to reproduce:
- Open the App, touch the "scan button" and scan n2-elite tag
- Select a slot on the n2 and hit activate
- Touch the tag to scan again
- The bank is changed, but the phone goes into a "tag-scan-loop" 
- Tag-Mo displays: "NFC not enabled"

- Restart fixes the issue

AbandonedCart, 2021-11-04:
I'm not able to reproduce it yet, but still trying.

choss, 2021-11-04:
What is funny is that it worked before. So I am really confused.

AbandonedCart, 2021-11-04:
I can't get it to break. I wonder if there is an issue with the scan related to some specific variable. I would normally say it was a case, but I scan through a Spigen Tough Armor with a PopWallet right where the NFC is located.

choss, 2021-11-04:
I use a POCO X3 NFC (stock case) and it did work before. I can feel the NFC buzzing directly after activating a new bank.
If I close the elite screen and rescan nothing works anymore. Maybe this helps. I will see if I am able to record a video.

AbandonedCart, 2021-11-04:
Hmm. Well, it definitely sounds like an NFC issue. It will probably require some sort of workaround. It is possible there are too many commands being passed around for the timeout on that hardware.

choss, 2021-11-04:
Have you increased the amount of commands in the last 5 days or so?

AbandonedCart, 2021-11-04:
There are a lot more verification steps. Some of them might be unnecessary. I'll see if I can slim it down a bit.

AbandonedCart, 2021-11-04:
The biggest problem (and reason for all the extra "bulk") is that regular, generic NFC cards / coins timeout in like 3 seconds. Any attempt to verify them at all causes them to come up as "lost" and halt the process. If I can get them to detect their type without breaking functionality, I can remove about 50% of the checks.

choss, 2021-11-04:
Thanks! That's all I can ask for. I'd even show you, if it were possible. 
In regards to that: Can you just detect if it is an n2 elite first and then do different things then?

AbandonedCart, 2021-11-04:
That was the original idea, which also broke the standard cards.

choss, 2021-11-04:
Because it is expected behaviour apparently. Normal tags should time out when issuing the "unlock" command (0x43) (e.g. https://lab401.com/products/undetectable-mifare-compatible-1k-one-time-write-uid )

My proposal would be: add a second can button for only scanning n2 elite to the screen (which is now *way* smaller) when n2 elite support is enabled. Skip the checks and do the (old) magic. 

I assume the users (both of them) which use n2 elite can live with that (I know I can)

AbandonedCart, 2021-11-04:
I know it will be a hassle for one person, but necessary is necessary.

AbandonedCart, 2021-11-04:
Come to think of it, I could probably do the inverse. If support is enabled, add an extra button for read / write of "normal" tags, since the assumption is most people with an N2 aren't using normal tags daily.

burndive, 2021-11-04:
Those of us with N2 Elite do in fact use normal tags, and one sort of tag in particular: actual amiibo. The N2 Elite is a great way to enjoy the benefits of amiibo while keeping the figures themselves on the shelf.

But personally I would prefer separate buttons to scan an N2 Elite vs. a regular amiibo/tag.

AbandonedCart, 2021-11-04:
TagMo cannot overwrite actual Amiibo.

choss, 2021-11-04:
I also tend to agree. Normally I don't use a real amiibo/NFC tag daily. 

For a scan of an amiibo for keeping it on the shelf the inverse sounds fine.  Have the scan "normal" tag somewhere else. 

And anyways: when bulk scanning/writing tags one can always toggle the option.

AbandonedCart, 2021-11-04:
Definitely not making separate apps. It's tedious trying to maintain one version without making sure they all build consistently and people download the right version(s). Besides, I think I figured out the issue.

In trying to register all the right components, the scanner was accepting scans from the browser screen. If anything was too close to the NFC while the window opened, it took the scan. Put simply, the browser was stealing scans from the actual scanner.

AbandonedCart, 2021-11-04:
I think the timeout unlock design explains a few lines of code that didn't quite make sense in the elite app. They would reconnect after checking the signature. That explains it. Maybe that is the key.

AbandonedCart, 2021-11-04:
Generic cards / coins and Amiibo will "double scan" and work without actually breaking anything else. N2 Elite and Power Tag devices will go right to work.

AbandonedCart, 2021-11-05:
Should be resolved with https://github.com/HiddenRamblings/TagMo/commit/e627e2c8b33608870191bb3220f3007ddd9414f9, but we can open this if it pops up again.

closed 2021-11-05 by AbandonedCart

choss, 2021-11-05:
Yes it works now! At least with the n2 :) Thanks for the efforts!