HiddenRamblings/TagMo #323: Slider is so random

Catflap62, 2021-11-04:
Trying to get the correct size with the slider for my phone is not working, the best setting I had was before the slider builds and I made the phone font one lower. For me personally I would prefer no slider at all.

AbandonedCart, 2021-11-04:
The checkbox disables it.

Catflap62, 2021-11-04:
I know it does, but now when try font one lower it does not affect the N2 view only the rest of Tagmo.

AbandonedCart, 2021-11-04:
Did you kill the app (recents, swipe it away) after disabling the checkbox?

For now, that is how to fully reset it. The rest of the code to dynamically reset isn't finished.

Catflap62, 2021-11-04:
Yes, I just really dislike the slider it affects all of Tagmo, I choose a number the view goes ridiculously small, I go into settings and that is ridiculously small as well, I make it bigger and it's just random. I can't get it to go one lower than it is when not using it.

AbandonedCart, 2021-11-04:
Alright. Best bet is to disable it and leave it that way.

burndive, 2021-11-04:
Even if I never enable the Scaling checkbox in Settings, as soon as I scan the N2 Elite, the interface is to small, and it propagates to every other interface in the app.

AbandonedCart, 2021-11-04:
The code you have installed is absolute. It cannot be changed until the entire apk is updated.

burndive, 2021-11-04:
I cleared the app data after installing the latest update, never checked the box to Scale N2 Browser View, and that is what I saw.

burndive, 2021-11-04:
Here's a screenshot after opening the app, scanning the N2, and pressing the back button to exit that interface, and then scrolling down half a screen.
[image]

burndive, 2021-11-04:
Oh, I thought you said it was already in the latest.

AbandonedCart, 2021-11-04:
Just like last time, the slider itself is the cause of issues. Enable scaling and you get 2.5 or disable it for stock views. If it doesn't immediately refresh the browser, swipe down to force a refresh. That is still touchy.

AbandonedCart, 2021-11-04:
Large screen - Original vs Scaled

[image]

[image]

AbandonedCart, 2021-11-04:
Small screen - Original vs Scaled

[image]

[image]

AbandonedCart, 2021-11-04:
Closing this out for removal of the slider

closed 2021-11-04 by AbandonedCart