HiddenRamblings/TagMo #324: Writing to a Tag stuck on Tag Detected

Hawxo, 2021-11-04:
Have previously been using without a problem this morning, 6 hours later when I come back to continue writing to the same brand of tags, I select a bin file, select write and I'm stuck on the tag detected screen with the app looking like it's stuck scanning the tag.

Worth noting the app advised me there was an update, updated, opened, advised another update available, updated and now I'm stuck in these endless tag detected loops when trying to write to a Tag now. 

Andriod 11 - Samsung S21 Ultra

AbandonedCart, 2021-11-04:
This should be fixed by the newest build. It will scan (and beep), scan again (and beep) and this will complete whatever task was being attempted with a regular tag.

Hawxo, 2021-11-04:
Ah fab, I rolled back to the old version and it was working fine so I'll update on a different device to check. Thanks for a swift response and the fix

AbandonedCart, 2021-11-05:
Merging https://github.com/HiddenRamblings/TagMo/issues/325 into this one for the next update to address both.

scriss312, 2021-11-05:
> Merging #325 into this one for the next update to address both.

Quick update: I enabled N2 Elite Support. The tags picked up faster but I still received the same message, even after clearing app cache and restarting my phone.

AbandonedCart, 2021-11-05:
Enabling N2 mode typically makes it slower and a reboot definitely won't help here. It's in the detection. https://github.com/HiddenRamblings/TagMo/commit/5ec015589340702af5366019b9eac5be9a11d491 should speed it up, but I'm not entirely thrilled about using the errors to determine how to respond. One change in the text and it breaks.

AbandonedCart, 2021-11-05:
https://github.com/HiddenRamblings/TagMo/commit/429c144f26c3d86f369da0bdb3b106ae762122ed should speed things up with N2 support disabled and better explain scanning with it.

scriss312, 2021-11-05:
I was able to read and write again after updating to the latest build. Process takes about 10 seconds with N2 enabled. Thank you for the quick fix!

TimeTravelPenguin, 2021-12-15:
I'm having this issue on the newest version

AbandonedCart, 2021-12-15:
> I'm having this issue on the newest version

Disable N2 support temporarily to write to normal tags. Everything that could be done to speed it up has already been done.

TimeTravelPenguin, 2021-12-20:
> > I'm having this issue on the newest version
> 
> Disable N2 support temporarily to write to normal tags. Everything that could be done to speed it up has already been done.

I don't even have it enabled

AbandonedCart, 2021-12-20:
I'm not sure it's the same issue, so I may need a log. The issue here was that N2 detection slowed regular tag scanning.

devonshirered, 2022-01-01:
Heya, I can't find what version I have but I found an update from 12 days ago and an experimental update from 4 days ago here, and neither have stopped the problem. I was writing fine a couple of days ago but now just stuck on the scanning/detected pages :( any help much appreciated! Thanks!

AbandonedCart, 2022-01-02:
Similar to the previous report, a log is needed.

blushnugget, 2022-01-02:
I'm having the same or a similar problem - stuck on "Mifare UltraLight C detected..." It's reading tags that I already have (that I didn't make) but that's as far as I get when I try to write new ones. I've tried the experimental release from a couple days ago and the one a few days before that, with the same problem. How would I send a log, if that helps?

AbandonedCart, 2022-01-02:
https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues

blushnugget, 2022-01-02:
> https://github.com/HiddenRamblings/TagMo/wiki#reporting-issues

Thank you! Found it. How do I send it to you?

AbandonedCart, 2022-01-02:
Should be able to drag and drop right here.

devonshirered, 2022-01-02:
samsung SM-G970F
Android SDK 30 (11)
TagMo f572858

--------- beginning of main
01-02 21:31:54.417 27911 27911 D InputTransport: Input channel constructed: '86938b4', fd=180
01-02 21:31:54.420 27911 27911 I ViewRootImpl@4786db[WebActivity]: setView = com.android.internal.policy.DecorView@cbd269d TM=true
01-02 21:31:54.478 27911 27911 I SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0x4ddf653 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
01-02 21:31:54.479 27911 27911 I ViewRootImpl@4786db[WebActivity]: Relayout returned: old=(0,117,1080,2136) new=(0,117,1080,2136) req=(1080,2019)0 dur=9 res=0x7 s={true 496164845312} ch=true fn=-1
01-02 21:31:54.483 27911 27911 I ViewRootImpl@4786db[WebActivity]: [DP] dp(1) 1 android.view.ViewRootImpl.reportNextDraw:10957 android.view.ViewRootImpl.performTraversals:3845 android.view.ViewRootImpl.doTraversal:2618 
01-02 21:31:54.483 27911 27911 I ViewRootImpl@4786db[WebActivity]: [DP] pd() Asnyc report
01-02 21:31:54.540 27911 27911 I ViewRootImpl@4786db[WebActivity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$performDraw$1$ViewRootImpl:4668 android.view.-$$Lambda$ViewRootImpl$DJd0VUYJgsebcnSohO6h8zc_ONI.run:6 android.os.Handler.handleCallback:938 
01-02 21:31:54.540 27911 27911 I ViewRootImpl@4786db[WebActivity]: [DP] rdf()
01-02 21:31:54.541 27911 27911 I ViewRootImpl@36231c0[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
01-02 21:31:54.550 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 145 0x7385a83850
01-02 21:31:54.551 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 146 0x7385a83850
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331916800]
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331916800]
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331926320]
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331926320]
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331916576]
01-02 21:31:54.561 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331916576]
01-02 21:31:54.567 27911 27911 I ViewRootImpl@4786db[WebActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
01-02 21:31:54.568 27911 27911 D InputMethodManager: prepareNavigationBarInfo() DecorView@cbd269d[WebActivity]
01-02 21:31:54.568 27911 27911 D InputMethodManager: getNavigationBarColor() -855310
01-02 21:31:54.568 27911 27911 D InputMethodManager: prepareNavigationBarInfo() DecorView@cbd269d[WebActivity]
01-02 21:31:54.568 27911 27911 D InputMethodManager: getNavigationBarColor() -855310
01-02 21:31:54.568 27911 27911 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
01-02 21:31:54.569 27911 27911 D InputMethodManager: startInputInner - Id : 0
01-02 21:31:54.569 27911 27911 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
01-02 21:31:54.571 27911 27911 D InputTransport: Input channel constructed: 'ClientS', fd=181
01-02 21:31:54.571 27911 27911 D InputTransport: Input channel destroyed: 'ClientS', fd=112
01-02 21:31:54.584 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331945360]
01-02 21:31:54.584 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331945360]
01-02 21:31:54.586 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331920944]
01-02 21:31:54.586 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331920944]
01-02 21:31:54.586 27911 27911 I SurfaceControl: nativeRelease nativeObject s[491331926992]
01-02 21:31:54.586 27911 27911 I SurfaceControl: nativeRelease nativeObject e[491331926992]
01-02 21:31:56.196 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:31:58.197 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:00.198 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:02.199 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:04.202 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:06.205 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:08.206 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:10.208 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:12.210 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:14.212 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-02 21:32:16.213 27911 27923 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
[... cut ...]
01-02 21:42:33.091 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 254 0x7385a83850
01-02 21:42:33.107 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 255 0x7385a83850
01-02 21:42:33.109 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 256 0x7385a83850
01-02 21:42:33.125 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 257 0x7385a83850
01-02 21:42:33.126 27911 27973 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 258 0x7385a83850

devonshirered, 2022-01-02:
Wont let me upload it as a file, sorry 😅

AbandonedCart, 2022-01-02:
`E ramblings.tagm: failed to connect to jdwp control socket: Connection refused` is saying that the app can't connect to the debugger to log anything. That's a bit odd, since it's not trying to perform a log that even requires permissions.

devonshirered, 2022-01-02:
I'll uninstall and reinstall again see what the new log says

AbandonedCart, 2022-01-02:
The issue may be in the firmware, not the app. From what I've read, some require enabling adb debugging for any logs to work. Just to be sure, I verified that I am still able to output a log with or without debugging enabled.

devonshirered, 2022-01-02:
samsung SM-G970F Android SDK 30 (11) TagMo f572858 --------- beginning of main 01-02 22:21:43.182 9759 9759 I ViewRootImpl@35e6e5c[PopupWindow:e08c7a9]: Relayout returned: old=(444,1344,1032,1776) new=(444,1344,1032,1776) req=(588,432)0 dur=4 res=0x1 s={true 496164566272} ch=false fn=23 01-02 22:21:43.251 9759 9759 I DecorView: [INFO] isPopOver=false, config=true 01-02 22:21:43.251 9759 9759 I DecorView: updateCaptionType >> DecorView@c6f873e[], isFloating=false, isApplication=true, hasWindowDecorCaption=false, hasWindowControllerCallback=true 01-02 22:21:43.251 9759 9759 D DecorView: setCaptionType = 0, this = DecorView@c6f873e[] 01-02 22:21:43.294 9759 9759 D InputTransport: Input channel constructed: 'b1edfe8', fd=103 01-02 22:21:43.296 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: setView = com.android.internal.policy.DecorView@c6f873e TM=true 01-02 22:21:43.297 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331932592] 01-02 22:21:43.297 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331932592] 01-02 22:21:43.308 9759 9759 I SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0x678637f / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1820 android.view.ViewRootImpl.relayoutWindow:9005 android.view.ViewRootImpl.performTraversals:3360 android.view.ViewRootImpl.doTraversal:2618 android.view.ViewRootImpl$TraversalRunnable.run:9971 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 01-02 22:21:43.309 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: Relayout returned: old=(0,0,1080,2280) new=(0,0,1080,2280) req=(1080,2280)0 dur=6 res=0x7 s={true 496164723232} ch=true fn=-1 01-02 22:21:43.312 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: [DP] dp(1) 1 android.view.ViewRootImpl.reportNextDraw:10957 android.view.ViewRootImpl.performTraversals:3845 android.view.ViewRootImpl.doTraversal:2618 01-02 22:21:43.314 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: [DP] pd() Asnyc report 01-02 22:21:43.324 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$performDraw$1$ViewRootImpl:4668 android.view.-$$Lambda$ViewRootImpl$DJd0VUYJgsebcnSohO6h8zc_ONI.run:6 android.os.Handler.handleCallback:938 01-02 22:21:43.324 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: [DP] rdf() 01-02 22:21:43.325 9759 9759 I ViewRootImpl@35e6e5c[PopupWindow:e08c7a9]: MSG_WINDOW_FOCUS_CHANGED 0 1 01-02 22:21:43.326 9759 9759 I ViewRootImpl@590661c[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1 01-02 22:21:43.327 9759 9759 I ViewRootImpl@590661c[BrowserActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331926208] 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331926208] 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331935504] 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331935504] 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331939200] 01-02 22:21:43.336 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331939200] 01-02 22:21:43.362 9759 9759 I ViewRootImpl@7badad9[NfcActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1 01-02 22:21:43.362 9759 9759 D InputMethodManager: prepareNavigationBarInfo() DecorView@c6f873e[NfcActivity] 01-02 22:21:43.362 9759 9759 D InputMethodManager: getNavigationBarColor() -855310 01-02 22:21:43.364 9759 9759 D EmergencyMode: [EmergencyManager] android createPackageContext successful: com.hiddenramblings.tagmo 01-02 22:21:43.370 9759 9759 D InputMethodManager: prepareNavigationBarInfo() DecorView@c6f873e[NfcActivity] 01-02 22:21:43.370 9759 9759 D InputMethodManager: getNavigationBarColor() -855310 01-02 22:21:43.370 9759 9759 V InputMethodManager: Starting input: tba=com.hiddenramblings.tagmo ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false 01-02 22:21:43.371 9759 9759 D InputMethodManager: startInputInner - Id : 0 01-02 22:21:43.371 9759 9759 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus 01-02 22:21:43.371 9759 9759 D InputTransport: Input channel constructed: 'ClientS', fd=131 01-02 22:21:43.371 9759 9759 D InputTransport: Input channel destroyed: 'ClientS', fd=81 01-02 22:21:43.396 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331898544] 01-02 22:21:43.396 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331898544] 01-02 22:21:43.397 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331929232] 01-02 22:21:43.397 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331929232] 01-02 22:21:43.397 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331933376] 01-02 22:21:43.397 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331933376] 01-02 22:21:43.497 9759 11268 W libEGL : EGLNativeWindowType 0x7385b67510 disconnect failed 01-02 22:21:43.497 9759 11268 D OpenGLRenderer: endAllActiveAnimators on 0x7315ad3560 (MenuPopupWindow$MenuDropDownListView) with handle 0x7225b05e10 01-02 22:21:43.498 9759 9759 D ViewRootImpl@35e6e5c[PopupWindow:e08c7a9]: dispatchDetachedFromWindow: reset blurmask 01-02 22:21:43.498 9759 9759 E SurfaceFlinger: resetPartialBlurMask failed to transact: -1 01-02 22:21:43.498 9759  [... cut ...] iveObject s[491331912096] 01-02 22:22:14.777 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331912096] 01-02 22:22:14.777 9759 9759 I SurfaceControl: nativeRelease nativeObject s[491331904032] 01-02 22:22:14.777 9759 9759 I SurfaceControl: nativeRelease nativeObject e[491331904032] 01-02 22:22:14.777 9759 9759 I ViewRootImpl@a4b90bc[PopupWindow:5973c89]: Relayout returned: old=(480,129,1068,705) new=(480,129,1068,705) req=(588,576)0 dur=8 res=0x1 s={true 496164836592} ch=false fn=22

AbandonedCart, 2022-01-02:
It still had a couple failed attempts to log, but enough got through to see an error. It doesn't seem to be in reading. It looks like it's happening when using the menu after loading a tag.

blushnugget, 2022-01-02:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7799977/tagmo_logcat.txt)

There's mine. When you say you've made a change, would we need to manually update the app or anything for it to apply? Really appreciate your help, by the way.

AbandonedCart, 2022-01-03:
When a new version is published, the app will ask to install it on the next launch. Unfortunately, the issue in that log is with the NFC hardware / drivers. That is something that can't be fixed in the app.

```
01-03 09:20:58.761 21757 21757 W IOException: java.io.IOException: Transceive failed
01-03 09:20:58.761 21757 21757 W IOException: 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:52)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.nfc.tech.NfcA.transceive(NfcA.java:123)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.readPages(NTAG215.java:76)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readFromTag(TagReader.java:58)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:357)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onNewIntent(NfcActivity.java:231)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.Activity.performNewIntent(Activity.java:8222)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1409)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1422)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4074)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4086)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:53)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2325)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.os.Handler.dispatchMessage(Handler.java:106)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.os.Looper.loop(Looper.java:246)
01-03 09:20:58.761 21757 21757 W IOException: 	at android.app.ActivityThread.main(ActivityThread.java:8633)
01-03 09:20:58.761 21757 21757 W IOException: 	at java.lang.reflect.Method.invoke(Native Method)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
01-03 09:20:58.761 21757 21757 W IOException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
01-03 09:20:58.762 21757 21757 I ViewRootImpl@d62c9bb[NfcActivity]: stopped(false) old=false
01-03 09:21:11.937 21757 21757 W IOException: java.io.IOException: Transceive failed
01-03 09:21:11.937 21757 21757 W IOException: 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:52)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.nfc.tech.NfcA.transceive(NfcA.java:123)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.readPages(NTAG215.java:76)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.nfctech.TagReader.readFromTag(TagReader.java:58)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:357)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.hiddenramblings.tagmo.NfcActivity.onNewIntent(NfcActivity.java:231)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.Activity.performNewIntent(Activity.java:8222)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1409)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1422)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4074)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4086)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:53)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2325)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.os.Handler.dispatchMessage(Handler.java:106)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.os.Looper.loop(Looper.java:246)
01-03 09:21:11.937 21757 21757 W IOException: 	at android.app.ActivityThread.main(ActivityThread.java:8633)
01-03 09:21:11.937 21757 21757 W IOException: 	at java.lang.reflect.Method.invoke(Native Method)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
01-03 09:21:11.937 21757 21757 W IOException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

blushnugget, 2022-01-03:
Ahh, okay. So the issue is my phone? Or the tags I'm using?

AbandonedCart, 2022-01-03:
It could be the tags, if they aren't NTAG215. If they are, then it is more likely the phone. I would assume the software over the hardware.

blushnugget, 2022-01-03:
I'll have to double check. Thanks so much for the help!

blushnugget, 2022-01-03:
It was the wrong type of tag. Hadn't realised there were different kinds - learning every day hahaha. Thanks.

AbandonedCart, 2022-01-03:
Glad you figured it out and it's unfortunate, but better than the alternative.

devonshirered, 2022-01-03:
Gone back to the bin files i used originally, updated to latest app when it asked me to, selected bones as ive already made him before so i know the bin file is fine, still stuck :(

samsung SM-G970F
Android SDK 30 (11)
TagMo f826b86

--------- beginning of main
01-03 16:14:38.694  3113  3113 I ViewRootImpl@24c29e8[PopupWindow:db41885]: MSG_WINDOW_FOCUS_CHANGED 1 1
01-03 16:14:38.704  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 305 0x7385aa1530
01-03 16:14:38.705  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 306 0x7385aa1530
01-03 16:14:38.722  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 307 0x7385aa1530
01-03 16:14:38.723  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 308 0x7385aa1530
01-03 16:14:38.738  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 309 0x7385aa1530
01-03 16:14:38.739  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 310 0x7385aa1530
01-03 16:14:38.755  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 311 0x7385aa1530
01-03 16:14:38.756  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 312 0x7385aa1530
01-03 16:14:38.774  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 313 0x7385aa1530
01-03 16:14:38.774  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 314 0x7385aa1530
01-03 16:14:38.942  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 315 0x7385aa1530
01-03 16:14:38.943  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 316 0x7385aa1530
01-03 16:14:38.957  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 317 0x7385aa1530
01-03 16:14:38.959  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 318 0x7385aa1530
01-03 16:14:38.972  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 319 0x7385aa1530
01-03 16:14:38.973  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 320 0x7385aa1530
01-03 16:14:38.991  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 321 0x7385aa1530
01-03 16:14:38.994  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 322 0x7385aa1530
01-03 16:14:39.007  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 323 0x7385aa1530
01-03 16:14:39.010  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 324 0x7385aa1530
01-03 16:14:39.025  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 325 0x7385aa1530
01-03 16:14:39.026  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 326 0x7385aa1530
01-03 16:14:39.038  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 327 0x7385aa1530
01-03 16:14:39.039  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 328 0x7385aa1530
01-03 16:14:39.058  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 329 0x7385aa1530
01-03 16:14:39.059  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 330 0x7385aa1530
01-03 16:14:39.073  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 331 0x7385aa1530
01-03 16:14:39.074  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 332 0x7385aa1530
01-03 16:14:39.088  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 333 0x7385aa1530
01-03 16:14:39.091  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 334 0x7385aa1530
01-03 16:14:39.104  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 335 0x7385aa1530
01-03 16:14:39.105  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 336 0x7385aa1530
01-03 16:14:39.126  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 337 0x7385aa1530
01-03 16:14:39.127  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 338 0x7385aa1530
01-03 16:14:39.143  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 339 0x7385aa1530
01-03 16:14:39.144  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 340 0x7385aa1530
01-03 16:14:39.158  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 341 0x7385aa1530
01-03 16:14:39.160  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 342 0x7385aa1530
01-03 16:14:39.174  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 343 0x7385aa1530
01-03 16:14:39.176  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 344 0x7385aa1530
01-03 16:14:39.195  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 345 0x7385aa1530
01-03 16:14:39.197  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 346 0x7385aa1530
01-03 16:14:39.210  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 347 0x7385aa1530
01-03 16:14:39.212  3113  5325 D RenderScript: [RS-DIAG] Launching GPU kernel intrinsic : id(5) 348 0x7385aa1530
01-03 16:14:39.301  3113  3113 I ViewRootImpl@24c29e8[PopupWindow:db41885]: ViewPostIme pointer 0
[... cut ...]
01-03 16:15:16.034  3113  3113 I SurfaceControl: nativeRelease nativeObject s[491331794272]
01-03 16:15:16.034  3113  3113 I SurfaceControl: nativeRelease nativeObject e[491331794272]
01-03 16:15:16.034  3113  3113 I ViewRootImpl@9ae32a0[PopupWindow:65a95d]: Relayout returned: old=(480,129,1068,705) new=(480,129,1068,705) req=(588,576)0 dur=11 res=0x1 s={true 496164784272} ch=false fn=21
01-03 16:15:16.039  3113  3113 I SurfaceControl: nativeRelease nativeObject s[491331897312]
01-03 16:15:16.039  3113  3113 I SurfaceControl: nativeRelease nativeObject e[491331897312]

devonshirered, 2022-01-03:
Same tags too

AbandonedCart, 2022-01-03:
```
01-03 16:14:46.618 3113 3113 W TagReader: 0477AA51171002898C480000E1103E000300FE000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000BD040000FF00050000
01-03 16:14:46.619 3113 3125 E ramblings.tagm: failed to connect to jdwp control socket: Connection refused
01-03 16:14:46.623 3113 3113 W TagWriter: Power Tag identified: false
01-03 16:14:46.636 3113 3113 W Exception: java.lang.Exception: Failed to decrypt tag data
01-03 16:14:46.636 3113 3113 W Exception: at com.hiddenramblings.tagmo.amiibo.KeyManager.decrypt(KeyManager.java:148)
01-03 16:14:46.636 3113 3113 W Exception: at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:87)
01-03 16:14:46.636 3113 3113 W Exception: at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:358)
01-03 16:14:46.636 3113 3113 W Exception: at com.hiddenramblings.tagmo.NfcActivity.onNewIntent(NfcActivity.java:231)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.Activity.performNewIntent(Activity.java:8222)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1409)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1422)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4074)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4086)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:53)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2325)
01-03 16:14:46.636 3113 3113 W Exception: at android.os.Handler.dispatchMessage(Handler.java:106)
01-03 16:14:46.636 3113 3113 W Exception: at android.os.Looper.loop(Looper.java:246)
01-03 16:14:46.636 3113 3113 W Exception: at android.app.ActivityThread.main(ActivityThread.java:8633)
01-03 16:14:46.636 3113 3113 W Exception: at java.lang.reflect.Method.invoke(Native Method)
01-03 16:14:46.636 3113 3113 W Exception: at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
01-03 16:14:46.636 3113 3113 W Exception: at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

```
01-03 16:15:09.420 3113 3113 W TagReader: 0477AA51171002898C480000E1103E000300FE000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000BD040000FF00050000
01-03 16:15:09.426 3113 3113 W TagWriter: Power Tag identified: false
01-03 16:15:09.439 3113 3113 W Exception: java.lang.Exception: Failed to decrypt tag data
01-03 16:15:09.439 3113 3113 W Exception: at com.hiddenramblings.tagmo.amiibo.KeyManager.decrypt(KeyManager.java:148)
01-03 16:15:09.439 3113 3113 W Exception: at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:87)
01-03 16:15:09.439 3113 3113 W Exception: at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:358)
01-03 16:15:09.439 3113 3113 W Exception: at com.hiddenramblings.tagmo.NfcActivity.onNewIntent(NfcActivity.java:231)
01-03 16:15:09.439 3113 3113 W Exception: at android.app.Activity.performNewIntent(Activity.java:8222)
[... cut ...]
01-03 16:15:09.439 3113 3113 W Exception: at java.lang.reflect.Method.invoke(Native Method)
01-03 16:15:09.439 3113 3113 W Exception: at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
```

It is still coming up that the tag itself can't be decrypted. "Normal" tags have a double scan when the N2 elite option is enabled, which was causing a few issues in the past. I just verified with both reading a tag and writing a new one, but got the correct results for both.

Are you using the separate keys (unfixed and locked) or a single key (retail)?

devonshirered, 2022-01-04:
Retail, I'll try with the unfixed and locked

devonshirered, 2022-01-04:
[image]

devonshirered, 2022-01-04:
Its workinggg 😍 thank you for your help

AbandonedCart, 2022-01-04:
Ill have to look into why that key isn't functioning anymore. It could have something to do with recent changes to improve detection or it may have been an unlucky copy.

closed 2022-01-04 by AbandonedCart

buwnga, 2025-01-19:
The error I am facing is

TagMo 4.3.3 (Play Release) #87054f42
motorola UPSIDE_DOWN_CAKE (14) - 14.00 GB RAM

java.lang.Exception: NTAG215 required, found NfcA!
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onMifareUltralight(NfcActivity.kt:515)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onMifareUltralight(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invokeSuspend(NfcActivity.kt:683)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onTagDiscovered$2.invoke(NfcActivity.kt:0)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:42)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:164)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.onTagDiscovered(NfcActivity.kt:675)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity.access$onTagDiscovered(NfcActivity.kt:52)
	at com.hiddenramblings.tagmo.nfctech.NfcActivity$onNewIntent$1.invokeSuspend(NfcActivity.kt:825)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:113)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:89)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:589)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:823)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:720)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:707)

buwnga, 2025-01-19:
It keeps showing ntag 215 required , nfcA found