HiddenRamblings/TagMo #325: Stuck writing and reading tags

scriss312, 2021-11-05:
On both current version and previous version, TagMo got stuck reading and writing tags. I keep seeing the message "Mifare UltraLight scanning..." But then nothing happens.
[image]

AWernerCS, 2021-11-05:
Same issue. I was using a much older version that worked just fine and updated tonight after getting a new batch of bin files, now I am unable to write - I get stuck on the exact same screen with the same message.

AbandonedCart, 2021-11-05:
Are you removing the tag when you hear a beep? Don't.

scriss312, 2021-11-05:
> Are you removing the tag when you hear a beep? Don't.

I tried both ways. It takes way longer to register a tag is even there compared to previous builds. And it doesn't seem to matter how long I leave the tag, I still have the same message. I've also tried clearing the app cache and restarting my phone and it's the same thing.

AbandonedCart, 2021-11-05:
It shouldn't take any longer if you haven't enabled N2 support. With that enabled, it does an extra scan to avoid corrupting your tags or N2.

AbandonedCart, 2021-11-05:
I'll look into speeding it up a bit more. There may be a loophole.

AbandonedCart, 2021-11-05:
Merged with the other open issue about scanning generic tags.

closed 2021-11-05 by AbandonedCart