HiddenRamblings/TagMo #326: Error while writing main data (stage 1)

Caelowo, 2021-11-05:
Hi! I'm getting this error while trying to write tags. The full error text is "Error while writing main data (stage 1) java.io.IOException: Trancieve failed.

Attaching screenshots from NXP tag info

[image]
[image]

[image]
[image]
[image]

AbandonedCart, 2021-11-05:
NXP appears to be saying they're write protected.

Caelowo, 2021-11-05:
> NXP appears to be saying they're write protected.

Hm. I've used these same tags before without issue... The tags that failed to write still scan in as amiibos when I scan them, but they don't work in animal crossing.

AbandonedCart, 2021-11-05:
I'll have to increase the logging and see if it gives a better reason. It looks like it is missing a write step. I wouldn't write any more tags until after the update, just to be safe. No use wasting them.

Caelowo, 2021-11-05:
Alright, thanks. I only have one empty tag left for now anyways, so I wouldn't want to use it until I'm more certain it would work. If it's any help I can show you the NXP tag info for the tags that got something written on them but don't work as amiibos?

AbandonedCart, 2021-11-05:
NXP info is practically useless for debugging why it didn't write. That just gives the status of the tag after it failed.

Caelowo, 2021-11-05:
Ah, okay. Well I can only wish you good luck then I guess, and give all of you working on the app my thanks!

AbandonedCart, 2021-11-05:
If https://github.com/HiddenRamblings/TagMo/commit/e3ffed6de00efc9ab5c2c8b60a5d63d0b6b1b3e9 doesn't end up fixing the issue, it will at least provide thorough logs

closed 2021-11-06 by AbandonedCart