HiddenRamblings/TagMo #327: Cant read Series 5 Animal Crossing Amiibo

NPW11, 2021-11-05:
I have downloaded the bin files for Series 5 Animal Crossing Amiibo but on tagmo they don't show up.

AbandonedCart, 2021-11-05:
Have you updated TagMo or synced the API? Made sure your keys are valid (because older ones don't work anymore)? Have you made sure you **backed up** your bins reliably?

NPW11, 2021-11-05:
[image]
I have them in the same folder as the other bin files and I have the updated tagmo.

NPW11, 2021-11-05:
> Have you updated TagMo or synced the API? Made sure your keys are valid (because older ones don't work anymore)? Have you made sure you **backed up** your bins reliably?

[image]
I have them in a folder with the other bin files and I have the updated tagmo.

NPW11, 2021-11-05:
> Have you updated TagMo or synced the API? Made sure your keys are valid (because older ones don't work anymore)? Have you made sure you **backed up** your bins reliably?

[image]

NPW11, 2021-11-05:
> > Have you updated TagMo or synced the API? Made sure your keys are valid (because older ones don't work anymore)? Have you made sure you **backed up** your bins reliably?
> 
> [image] I have them in a folder with the other bin files and I have the updated tagmo.

[image]

NPW11, 2021-11-05:
> Have you updated TagMo or synced the API? Made sure your keys are valid (because older ones don't work anymore)? Have you made sure you **backed up** your bins reliably?

I have them in a folder with the other bin files and I have the updated tagmo
[image]
[image]

AbandonedCart, 2021-11-05:
That doesn't answer any of the questions, though.

Have you updated the TagMo app or synced the API in settings?

After doing one of the above, did you refresh in the browser?

Are you sure the files are compatible with TagMo and your keys?

NPW11, 2021-11-05:
> That doesn't answer any of the questions, though.
> 
> Have you updated the TagMo app or synced the API in settings?
> 
> After doing one of the above, did you refresh in the browser?
> 
> Are you sure the files are compatible with TagMo and your keys?

I have updated tagmo and I have synced the API. I have refreshed. They should be, I downloaded them from the same place as the other files and they work so I'm not sure why series 5 isn't working

AbandonedCart, 2021-11-05:
We can't provide support for pirated files. Sorry.

closed 2021-11-05 by AbandonedCart