HiddenRamblings/TagMo #329: N2: No re-scan for same type of amiibo

burndive, 2021-11-05:
If two different instances of the same type of amiibo (example: two SSB Zelda amiibo with different data stored on them) are loaded to two different banks, whichever is scanned first when "Edit Properties" will be presented as the data for the second as well when "Edit Properties" is selected (the second bank is never scanned).

AbandonedCart, 2021-11-05:
I wouldn't even know where to start debugging this, since the data is assigned to the instance and not the Amiibo object.

burndive, 2021-11-05:
I'm not sure what exactly you mean by "instance" and "Amiibo object".

How to reproduce:

1. Save the same bin to two different slots, for example, banks 1 and 2
2. On bank : Edit Properties -> Generate Random Serial, change Nickname or other attributes -> Save -> Write
3. Exit to the main screen
4. Re-scan N2 Elite
5. Open bank 1, tap "Edit Properties", Scan, and observe the data that you changed
6. Open bank 2, tap "Edit Properties" **(There will be no scan)**. The data will be bank 1's data, not the original unmodified bank 2 data.
7. Exit to the main screen
8. Re-scan N2 Elite
9. Open bank 2, tap "Edit Properties", **Scan**, and observe the original unmodified bank 2 data.

AbandonedCart, 2021-11-05:
An instance is the specific item of a given type or object, which is an Amiibo in this case. A collection of objects could be searched by a parameter to get the closest match. The instance is finite. It is the single item with the exact parameters.

... list is made and displayed in a given order...

```
    public void onAmiiboClicked(Amiibo amiibo, int position) {
        ...
        clickedPosition = position; <- the position of the item, or instance reference
        ...
    }
```

```
 private void scanAmiiboData(int current_bank) {
        ...
        onScanTagResult.launch(scan);
    }
```

```
...onScanTagResult...
        byte[] tagData = result.getData().getByteArrayExtra(TagMo.EXTRA_TAG_DATA);
        if (amiibos.get(clickedPosition) != null) <- checking that the instance exists
            amiibos.get(clickedPosition).data = tagData; <- data stored to the instance
       ...
```

```
       ...
       editor.putExtra(TagMo.EXTRA_TAG_DATA, tagData); <- the same data as above
       onUpdateTagResult.launch(editor); 
```

```
      ... onUpdateTagResult...
       if (result.getData().hasExtra(TagMo.EXTRA_TAG_DATA)) {
            byte[] tagData = result.getData().getByteArrayExtra(TagMo.EXTRA_TAG_DATA); <- data from the editor
            if (amiibos.get(clickedPosition) != null) <- checking that the instance exists
                     amiibos.get(clickedPosition).data = tagData; <- the same instance as above updated
            updateAmiiboView(tagData, clickedPosition); <- data from editor and instance reference
       }
```

There is nowhere that a second, completely different, Amiibo is being confused.

burndive, 2021-11-05:
I think my use of the word "instance" in its more generic English sense rather than it's specific object oriented programming sense may have been confusing. 

Here's a video where I view the data on Bank 1 and Bank 2, both of which appear to have "Squid 1" in them, and then back out to re-scan the N2 and show that Bank 2 actually has "Squid 2".

https://youtu.be/2OXKSHsSQSM

The point of failure is where TagMo does not initiate a scan of Bank 2, since it thinks (incorrectly) that it already has that data.

AbandonedCart, 2021-11-06:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/a189db367f940c262b6f0049632b22a5cb3d3a53

closed 2021-11-06 by AbandonedCart

burndive, 2021-11-08:
This issue is still unfixed in f44dc00.

I have 2 different instances of Inkling Squid on banks 1 and 2.
If I select "Edit Properties" on bank 1 first, and then on bank 2, it skips the scan and shows me bank 1 data for bank 2.
If I select "Edit Properties" on bank 2 first, and then on bank 1, it skips the scan and shows me bank 2 data for bank 1.