HiddenRamblings/TagMo #332: Attempting to use the update function causes a crash

L31FY, 2021-11-07:
samsung SM-G998U1
Android SDK 31 (12)
TagMo Version 2.7.5

--------- beginning of main
11-06 19:46:19.863  6266  6266 E ramblings.tagm: Not starting debugger since process cannot load the jdwp agent.
11-06 19:46:19.864  6266  6266 E USNET   : USNET: appName: com.hiddenramblings.tagmo
11-06 19:46:19.865  6266  6266 D ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
11-06 19:46:19.867  6266  6266 D ActivityThread: setConscryptValidator
11-06 19:46:19.867  6266  6266 D ActivityThread: setConscryptValidator - put
--------- beginning of system
11-06 19:46:19.883  6266  6266 D ActivityThread: handleBindApplication()++ app=com.hiddenramblings.tagmo
11-06 19:46:19.883  6266  6266 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 11166; state: ENABLED
11-06 19:46:19.901  6266  6266 V GraphicsEnvironment: ANGLE Developer option for 'com.hiddenramblings.tagmo' set to: 'default'
11-06 19:46:19.901  6266  6266 V GraphicsEnvironment: App is not on the allowlist for updatable production driver.
11-06 19:46:19.902  6266  6266 D LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=com.hiddenramblings.tagmo appContext.mBasePackageName=com.hiddenramblings.tagmo
11-06 19:46:19.902  6266  6266 D NetworkSecurityConfig: No Network Security Config specified, using platform default
11-06 19:46:19.903  6266  6266 D NetworkSecurityConfig: No Network Security Config specified, using platform default
11-06 19:46:19.905  6266  6286 E ramblings.tagm: Invalid ID 0x00000000.
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: An exception occurred : 
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: android.content.res.Resources$NotFoundException: Resource ID #0x0
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: 	at android.content.res.ResourcesImpl.getValueForDensity(ResourcesImpl.java:249)
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: 	at android.content.res.Resources.offloadDrawable(Resources.java:2826)
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: 	at android.app.ResourcesOffloading$ResourcesOffloadingRunnable.run(ResourcesOffloading.java:82)
11-06 19:46:19.905  6266  6286 W ResourcesOffloading: 	at java.lang.Thread.run(Thread.java:920)
11-06 19:46:19.907  6266  6266 D ActivityThread: handleBindApplication() --
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: QUALCOMM build                   : b25f720, I2fb6e682e6
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Build Date                       : 09/14/21
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: OpenGL ES Shader Compiler Version: EV031.35.01.09
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Local Branch                     : CLDebug0914
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Remote Branch                    : 
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Remote Branch                    : 
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Reconstruct Branch               : 
11-06 19:46:19.955  6266  6289 I AdrenoGLES-0: Build Config                     : S P 10.0.7 AArch64
11-06 19:46:19.956  6266  6289 I AdrenoGLES-0: Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
11-06 19:46:19.958  6266  6289 I AdrenoGLES-0: PFP: 0x016dc094, ME: 0x00000000
11-06 19:46:19.970  6266  6266 I DecorView: [INFO] isPopOver=false, config=true
11-06 19:46:19.970  6266  6266 I DecorView: updateCaptionType >> DecorView@299743e[], isFloating=false, isApplication=true, hasWindowControllerCallback=true, hasWindowDecorCaption=false
11-06 19:46:19.970  6266  6266 D DecorView: setCaptionType = 0, this = DecorView@299743e[]
11-06 19:46:19.991  6266  6266 I MSHandlerLifeCycle: isMultiSplitHandlerRequested: ignored. pkg=com.hiddenramblings.tagmo parent=null callers=com.android.internal.policy.DecorView.setVisibility:4216 android.app.ActivityThread.handleResumeActivity:5285 android.app.servertransaction.ResumeActivityItem.execute:54 android.app.servertransaction.ActivityTransactionItem.execute:45 android.app.servertransaction.TransactionExecutor.executeLifecycleState:176 
11-06 19:46:19.991  6266  6266 I MSHandlerLifeCycle: removeMultiSplitHandler: no exist. decor=DecorView@299743e[]
11-06 19:46:19.995  6266  6289 D NativeCustomFrequencyManager: [NativeCFMS] BpCustomFrequencyManager::BpCustomFrequencyManager()
11-06 19:46:20.001  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: setView = com.android.internal.policy.DecorView@299743e TM=true
11-06 19:46:20.001  6266  6266 I MSHandlerLifeCycle: isMultiSplitHandlerRequested: windowingMode=1 isFullscreen=true isPopOver=false isHidden=false skipActivityType=false isHandlerType=true this: DecorView@299743e[BrowserActivity_]
11-06 19:46:20.001  6266  6266 I MSHandlerLifeCycle: removeMultiSplitHandler: no exist. decor=DecorView@299743e[BrowserActivity_]
11-06 19:46:20.015  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: Relayout returned: old=(0,0,1440,3200) new=(0,0,1440,3200) req=(1440,3200)0 dur=6 res=0x7 s={true 531905075008} ch=true fn=-1
11-06 19:46:20.017  6266  6289 D hw-ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
11-06 19:46:20.017  6266  6302 E SchedPolicy: Failed to find cgroup for tid 6266
11-06 19:46:20.067  6266  6289 I BufferQueueProducer: [ViewRootImpl@f592519[BrowserActivity_]#0(BLAST Consumer)0](id:187a00000000,api:1,p:6266,c:6266) queueBuffer: queued for the first time.
11-06 19:46:20.069  6266  6289 D OpenGLRenderer: GPIS:: SetUp Pid : 6266    Tid : 6289
11-06 19:46:20.107  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: MSG_WINDOW_FOCUS_CHANGED 1 1
11-06 19:46:20.108  6266  6266 D InputMethodManager: startInputInner - Id : 0
11-06 19:46:20.108  6266  6266 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
11-06 19:46:20.112  6266  6266 D InputMethodManager: startInputInner - Id : 0
11-06 19:46:21.877  6266  6277 W System  : A resource failed to call end. 
11-06 19:46:22.058  6266  6266 W Glide   : Failed to find GeneratedAppGlideModule. You should include an annotationProcessor compile dependency on com.github.bumptech.glide:compiler in your application and a @GlideModule annotated AppGlideModule implementation or LibraryGlideModules will be silently ignored
11-06 19:46:26.454  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: MSG_WINDOW_FOCUS_CHANGED 0 1
11-06 19:46:27.418  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: MSG_WINDOW_FOCUS_CHANGED 1 1
11-06 19:46:27.420  6266  6266 D InputMethodManager: startInputInner - Id : 0
11-06 19:46:27.420  6266  6266 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
11-06 19:46:31.213  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 0
11-06 19:46:31.619  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 1
11-06 19:46:33.020  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 0
11-06 19:46:33.867  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 1
11-06 19:46:35.983  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 0
11-06 19:46:36.048  6266  6266 I ViewRootImpl@f592519[BrowserActivity_]: ViewPostIme pointer 1
11-06 19:46:36.088  6266  6289 D NativeCustomFrequencyManager: [NativeCFMS] BpCustomFrequencyManager::BpCustomFrequencyManager()
11-06 19:46:36.096  6266  6266 I ViewRootImpl@4a13dac[PopupWindow:27a3bb9]: setView = android.widget.PopupWindow$PopupDecorView@879da75 TM=true
11-06 19:46:36.105  6266  6266 I ViewRootImpl@4a13dac[PopupWindow:27a3bb9]: Relayout returned: old=(0,100,1440,3020) new=(690,115,1425,1375) req=(735,1260)0 dur=4 res=0x7 s={true 531905057568} ch=true fn=-1
11-06 19:46:36.115  6266  6266 D AbsListView:  in onLayout changed 
[... 472 lines cut ...]
11-06 19:47:57.511  8963  8963 I ViewRootImpl@17806b6[BrowserActivity_]: MSG_WINDOW_FOCUS_CHANGED 1 1
11-06 19:47:57.511  8963  8963 D InputMethodManager: startInputInner - Id : 0
11-06 19:47:57.511  8963  8963 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
11-06 19:47:57.527  8963  8963 D InputMethodManager: startInputInner - Id : 0
11-06 19:47:58.588  8963  8963 I ViewRootImpl@17806b6[BrowserActivity_]: ViewPostIme pointer 0
11-06 19:47:58.645  8963  8963 I ViewRootImpl@17806b6[BrowserActivity_]: ViewPostIme pointer 1
11-06 19:47:58.663  8963  9035 D NativeCustomFrequencyManager: [NativeCFMS] BpCustomFrequencyManager::BpCustomFrequencyManager()
[... cut ...]
11-06 19:47:59.584  8963  8963 I ViewRootImpl@707a22b[PopupWindow:7dc0c9c]: Relayout returned: old=(690,115,1425,1015) new=(690,115,1425,1015) req=(735,900)0 dur=8 res=0x1 s={true 531905066288} ch=false fn=5
11-06 19:47:59.603  8963  8963 I ViewRootImpl@707a22b[PopupWindow:7dc0c9c]: dispatchDetachedFromWindow
11-06 19:47:59.605  8963  8963 D InputTransport: Input channel destroyed: '1b67649', fd=133
11-06 19:47:59.607  8963  8963 I ViewRootImpl@17806b6[BrowserActivity_]: MSG_WINDOW_FOCUS_CHANGED 1 1
11-06 19:47:59.607  8963  8963 D InputMethodManager: startInputInner - Id : 0
11-06 19:47:59.607  8963  8963 I InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus

AbandonedCart, 2021-11-07:
I grabbed the log and removed the content for your privacy. The log is only supposed to pick up "TagMo" content, but sometimes includes NFC data that I prefer to filter out just in case.

AbandonedCart, 2021-11-07:
The good news is that one of the issues was recently solved. The bad news is that the update issue doesn't appear to be included in the logs. There have been some improvements to that, though, that may have fixed it. You may have to perform a manual update to jump-start the fix. Give me a minute to look into that first error, though. That is one I haven't seen before.

L31FY, 2021-11-07:
That is a more than acceptable fix. I hate to keep blaming it, but it's not completely unlikely that Android 12 Beta over here has a relationship with it. I've had more breakage and outright strange issues with everything.

AbandonedCart, 2021-11-07:
They changed a lot and the new Android Emulator requires upgrading to a Mac version that is equally broken, so it's been a struggle to debug Android 12.

closed 2021-11-07 by AbandonedCart