HiddenRamblings/TagMo #334: Support unencrypted (generated) files

AbandonedCart, 2021-11-07:
Came across https://hax0kartik.github.io/amiibo-generator/ and got the idea that it may be not only possible, but preferable to support generated bin files.

This still requires some research, but will not replace existing support for backups.

AbandonedCart, 2021-11-07:
This actually ended up being super easy. We could technically ship with support for downloading these directly into the app. No bin files required. Only keys.

AbandonedCart, 2021-11-08:
* Download the latest experimental build.
* Open the app settings and scroll to the very bottom.
* Wait for the link to appear and click bulk download.
* Wait for the download to finish and press back.
* Watch as the dialog throws up lots of text.
* Return to the browser and select the new folder.

ishtangli, 2021-11-08:
Tried this but tags didnt show up. Promising though

closed 2021-11-08 by AbandonedCart