HiddenRamblings/TagMo #335: Can't select encryption keys

Natetop, 2021-11-08:
Google Pixel 3a Android SDK 31 (12) TagMo Version 2.7.5 --------- beginning of main 11-07 21:29:00.441 25588 25588 D CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10484; state: ENABLED 11-07 21:29:04.336 25588 25588 W Glide : Failed to find GeneratedAppGlideModule. You should include an annotationProcessor compile dependency on com.github.bumptech.glide:compiler in your application and a @GlideModule annotated AppGlideModule implementation or LibraryGlideModules will be silently ignored 11-07 21:29:08.937 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ee09690 (RippleDrawable) with handle 0x709eda3fd0 11-07 21:29:08.937 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718edd7850 (n) with handle 0x709eda1990 11-07 21:29:08.937 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718edf6800 (CardView) with handle 0x709eda0100 11-07 21:29:11.379 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ee24c50 (RippleDrawable) with handle 0x709eda77b0 11-07 21:30:00.433 25588 25674 E NumberFormatException: java.lang.NumberFormatException: For input string: "msf:126661" 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.lang.Long.parseLong(Long.java:594) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.lang.Long.valueOf(Long.java:808) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at a2.b.b(Unknown Source:84) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at t3.a.e(Unknown Source:4) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at w3.i.p3(Unknown Source:2) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at w3.l.x3(Unknown Source:0) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at w3.l$t.j(Unknown Source:4) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at h4.a$b.run(Unknown Source:19) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 11-07 21:30:00.433 25588 25674 E NumberFormatException: 	at java.lang.Thread.run(Thread.java:920) 11-07 21:31:20.697 25588 25588 E ClassCastException: java.lang.ClassCastException: java.lang.String cannot be cast to org.json.JSONObject 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at w3.i.B2(Unknown Source:9) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at w3.i.o2(Unknown Source:0) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at w3.h.a(Unknown Source:2) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at s3.a.c(Unknown Source:7) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at s3.a.onPostExecute(Unknown Source:2) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.AsyncTask.finish(AsyncTask.java:771) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.AsyncTask.access$900(AsyncTask.java:199) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:788) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.Handler.dispatchMessage(Handler.java:106) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.Looper.loopOnce(Looper.java:201) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.os.Looper.loop(Looper.java:288) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at android.app.ActivityThread.main(ActivityThread.java:7842) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at java.lang.reflect.Method.invoke(Native Method) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 11-07 21:31:20.697 25588 25588 E ClassCastException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003) 11-07 21:31:21.005 25588 25588 E ClassCastException: java.lang.ClassCastException: java.lang.String cannot be cast to org.json.JSONObject 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at w3.i.B2(Unknown Source:9) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at w3.i.o2(Unknown Source:0) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at w3.h.a(Unknown Source:2) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at s3.a.c(Unknown Source:7) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at s3.a.onPostExecute(Unknown Source:2) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.AsyncTask.finish(AsyncTask.java:771) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.AsyncTask.access$900(AsyncTask.java:199) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:788) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.Handler.dispatchMessage(Handler.java:106) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.Looper.loopOnce(Looper.java:201) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.os.Looper.loop(Looper.java:288) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at android.app.ActivityThread.main(ActivityThread.java:7842) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at java.lang.reflect.Method.invoke(Native Method) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 11-07 21:31:21.005 25588 25588 E ClassCastException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003) 11-07 21:31:32.993 25588 29565 D MediaScannerConnection: Scanned /storage/emulated/0/Android/data/com.hiddenramblings.tagmo/files/tagmo_logcat.txt to null 11-07 21:31:33.227 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ee814f0 (k0$a) with handle 0x709ed77720 11-07 21:32:09.519 25588 29565 D MediaScannerConnection: Scanned /storage/emulated/0/Android/data/com.hiddenramblings.tagmo/files/tagmo_logcat.txt to null 11-07 21:32:09.761 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718eeb4680 (k0$a) with handle 0x709edab6e0 11-07 21:32:21.566 25588 25588 D CompatibilityChangeReporter: Compat change id reported: 160794467; UID 10484; state: ENABLED 11-07 21:32:22.046 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ed38ff0 (RippleDrawable) with handle 0x709edac550 11-07 21:32:22.062 25588 25588 W Choreographer: Frame time is 0.219016 ms in the future! Check that graphics HAL is generating vsync timestamps using the correct timebase. 11-07 21:32:25.540 25588 25852 D h : 04C01955EAD16480DF480FE0F110FFEEA50000003B393B78AEA7643FCFE87E1CE39EDC0E3723030EB94E5822B8FFA595F69FB02D2E5FFEF8247494A5139406A46C9DF05779A854E816C09E72AFFBFB756D53D1260A0F000103C60502383939316031656390EEEA78B3919AEDDFE1A31C40337FC7FBF58B7F8CBE25085211AA32A1031FBF22301B5A41F5AB8EE9C69324270DEDD13AB84284E81BBF5B5CB72314B5C6D6EBEEB0B9F6D7AB95E730E94E6DD040E23413229AA7D214A2431F50CE9EA8A81D92BF44E85AAE73BFBF0CE0F2224340698CFC32176A3BD8F55FC3B5B31E7B3B647D7DDDDC8EB493AED8319A6A45D6B07B36DC0CD3E60BA19696E22C2CDF4FE3333A9F47B3A2A8ACD5F93DB4502BEEE142E22B088033DF69F5D28391FA00FD400B75EC0B20475B95D5EFFEDC5980101C963417AB969CC07FAA6D1774F6F8B805C924DF5D14154571E1C78297889875FB9C5683DBF0E80CFEA160A7604C7D1A6E47C7E18E9DFAB220141A26102EB6DFB63F1A43D4B9789C3DC078374648D1137B47BD73AC5AC80573E04B351DFF74895EB8F24F02663B408A10723431719C45120A770E82BCA7ED8545F906DBB8E323BC7BFEB9D8E0A382D1A61D9A1723AD91C735BC26DC17565249989631C91341447339AB08ABFFDDD39E497E6FDDB748B94698B8F12EB0E1223FD6F10A9684BA870E641DDE641DCBAEA73E00668342198ED94B60CB9FEBE601000FBD000000045F000000 11-07 21:32:36.048 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ed63a50 (RippleDrawable) with handle 0x709edb5430 11-07 21:32:36.784 25588 25588 E ClassCastException: java.lang.ClassCastException: java.lang.String cannot be cast to org.json.JSONObject 11-07 21:32:36.784 25588 25588 E ClassCastException: 	at w3.i.B2(Unknown Source:9) 11-07 21:32:36.784 25588 25588 E ClassCastExc [... cut ...] :53.909 25588 29510 E NumberFormatException: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 11-07 21:32:53.909 25588 29510 E NumberFormatException: 	at java.lang.Thread.run(Thread.java:920) 11-07 21:33:02.250 25588 29565 D MediaScannerConnection: Scanned /storage/emulated/0/Android/data/com.hiddenramblings.tagmo/files/tagmo_logcat.txt to null 11-07 21:33:02.561 25588 25609 D OpenGLRenderer: endAllActiveAnimators on 0x718ee91ce0 (k0$a) with handle 0x709eda1c90

zenxtang, 2021-11-08:
I am not sure if my issue is a duplicate of this or a new one issue, but my unfixed key loads but my locked key doesn't. They are both in key_retail.bin.

[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7494533/tagmo_logcat.txt)

Natetop, 2021-11-08:
> Hopefully https://github.com/HiddenRamblings/TagMo/releases/download/experimental/TagMo-9393209-signed.apk fixes it.

Yeah that seemed to fix it for me

closed 2021-11-08 by Natetop

qwertykid, 2023-03-31:
I'm having the same issue but the apk have a 404.

AbandonedCart, 2023-03-31:
https://github.com/HiddenRamblings/TagMo/releases