HiddenRamblings/TagMo #336: Purple text and view missing

ishtangli, 2021-11-08:
Some Amiibos have a purple text.. and it moves when you refresh. What are the purple ones?

[image]

Also, the count below shows i have missing amiibos  any way to see which they are?

AbandonedCart, 2021-11-08:
Extended menu -> Content -> Show Missing

ishtangli, 2021-11-08:
Tried it and it just says searching amiibo. Left it for a while too

AbandonedCart, 2021-11-08:
Sounds like your loading time is just really slow. What device is it?

ishtangli, 2021-11-10:
Oneplus 6T McLaren

AbandonedCart, 2021-11-10:
https://github.com/HiddenRamblings/TagMo/commit/00c144c4af567c17c6714ef7452fcce9509ce31b should help, at least a little.

AbandonedCart, 2021-11-11:
If you have the option to show missing enabled, the path is highlighted for ones you have as they appear. This was implemented before me, so I will have to look into it. 

I will also disable the search text with the option enabled, since even not finding files will still have results. It was set to display until files are found, but showing missing breaks the check.

closed 2021-11-11 by AbandonedCart