HiddenRamblings/TagMo #337: Keys won't load in latest release

Muchieman7, 2021-11-08:
Was the latest release meant to be published? It doesn't even have a proper version number

AbandonedCart, 2021-11-08:
https://github.com/HiddenRamblings/TagMo/wiki/Installing-TagMo

Your issue is fixed in the experimental branch. If you are experiencing the greyed out selection, it is because the keys are in the root of the downloads folder. That hasn't been addressed, but a workaround is to move them into a subfolder or somewhere else.

Next time you post a bug, such as not being able to access a file, a log is much more useful than complaining about the version numbering.

Muchieman7, 2021-11-08:
Tell me how to post a log file if you need it :) I wasn't complaining, I'm just on mobile and am struggling to report a bug at all. It worked fine in 2.7.5, so I'd figure I'd report the issue just in case.

AbandonedCart, 2021-11-08:
In the extended menu of the main browser is a logcat option. The best time to get one is right after trying to do whatever doesn't work.

AbandonedCart, 2021-11-09:
Some changes were made for this. Testing now.

Muchieman7, 2021-11-09:
```
Google Pixel 4 XL
Android SDK 31 (12)
TagMo Version 2.7.5

--------- beginning of main
11-08 19:02:37.273 23742 23742 D CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10261; state: ENABLED
11-08 19:03:01.183 23742 23742 W ramblings.tagm: Accessing hidden field Landroid/widget/AbsListView;->mIsChildViewEnabled:Z (unsupported, reflection, allowed)
11-08 19:03:03.532 23742 24363 D MediaScannerConnection: Scanned /storage/emulated/0/Android/data/com.hiddenramblings.tagmo/files/tagmo_logcat.txt to null
11-08 19:03:03.752 23742 23763 D OpenGLRenderer: endAllActiveAnimators on 0xb40000768c5624b0 (k0$a) with handle 0xb40000759c54e480
11-08 19:03:38.844 23742 23756 W System  : A resource failed to call end. 
11-08 19:04:07.118 23742 24363 D MediaScannerConnection: Scanned /storage/emulated/0/Android/data/com.hiddenramblings.tagmo/files/tagmo_logcat.txt to null
11-08 19:04:07.367 23742 23763 D OpenGLRenderer: endAllActiveAnimators on 0xb40000768c596320 (k0$a) with handle 0xb40000759c533b40
11-08 19:05:28.392 23742 23763 D OpenGLRenderer: endAllActiveAnimators on 0xb40000768c583490 (RippleDrawable) with handle 0xb40000759c54d550
11-08 19:05:29.127 23742 23742 E ClassCastException: java.lang.ClassCastException: java.lang.String cannot be cast to org.json.JSONObject
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at v3.i.B2(:140)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at v3.i.o2(Unknown Source:0)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at v3.h.a(Unknown Source:2)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at r3.a.c(:37)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at r3.a.onPostExecute(:15)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.AsyncTask.finish(AsyncTask.java:771)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.AsyncTask.access$900(AsyncTask.java:199)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:788)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.Looper.loopOnce(Looper.java:201)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.os.Looper.loop(Looper.java:288)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at android.app.ActivityThread.main(ActivityThread.java:7842)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at java.lang.reflect.Method.invoke(Native Method)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
11-08 19:05:29.127 23742 23742 E ClassCastException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
11-08 19:05:29.521 23742 23763 D OpenGLRenderer: endAllActiveAnimators on 0xb40000768c523870 (RippleDrawable) with handle 0xb40000759c557cc0
11-08 19:05:36.498 23742 23742 E ClassCastException: java.lang.ClassCastException: java.lang.String cannot be cast to org.json.JSONObject
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at v3.i.B2(:140)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at v3.i.o2(Unknown Source:0)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at v3.h.a(Unknown Source:2)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at r3.a.c(:37)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at r3.a.onPostExecute(:15)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.AsyncTask.finish(AsyncTask.java:771)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.AsyncTask.access$900(AsyncTask.java:199)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.AsyncTask$InternalHandler.handleMessage(AsyncTask.java:788)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.Looper.loopOnce(Looper.java:201)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.os.Looper.loop(Looper.java:288)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at android.app.ActivityThread.main(ActivityThread.java:7842)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at java.lang.reflect.Method.invoke(Native Method)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
11-08 19:05:36.498 23742 23742 E ClassCastException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
11-08 19:05:36.725 23742 23763 D OpenGLRenderer: endAllActiveAnimators on 0xb40000768c583490 (RippleDrawable) with handle 0xb40000759c5335a0
11-08 19:05:37.280 23742 23742 W Choreographer: Frame time is 0.027177 ms in the future!  Check that graphics HAL is generating vsync timestamps using the correct timebase.
```

AbandonedCart, 2021-11-09:
That looks like an older build. AsyncTask has been removed.

closed 2021-11-10 by AbandonedCart