HiddenRamblings/TagMo #339: Build Wumiibo error page

x7l7j8cc, 2021-11-09:
As the title suggest, when i tap on the Buil Wumiibo option i get this error page.
And the decrypted amiibo doesnt show up anymore, while becore the last update worked perfectly.

[image]

AbandonedCart, 2021-11-09:
Make sure to check the box to display the download folder. It's in the extended menu under content.

x7l7j8cc, 2021-11-09:
It's checked, but nothing.
Decrypted Amiibo doesnt show up, and the Build Wumiibo page give that error.

But as i said, yesterday before the latest update (the version with 'Amiibo Generator' page) worked.

AbandonedCart, 2021-11-09:
Yes. There were some changes since then. It appears not all phones are compatible with them.

x7l7j8cc, 2021-11-09:
Can you send me the version with the 'Amiibo Generator' page?
At least that worked fine for me.

Anyway, i have a huawei p30 pro, if this helps for something. And if you want i can test for you a version that fix the problem on this phone (dont know others), if you want.

AbandonedCart, 2021-11-09:
I don't have that old version. They are replaced when a new one builds.

x7l7j8cc, 2021-11-09:
Great, no decrypted amiibo (need to recrypt, again)...
Anyway, neither on my Samsung S6 Tablet work that page.

[image]

AbandonedCart, 2021-11-09:
I believed you.

AbandonedCart, 2021-11-09:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/3e5a31ebb13aa24f81c8acbb21996192d7279552

closed 2021-11-09 by AbandonedCart

x7l7j8cc, 2021-11-09:
Yep, thhis new version , fixed the error on both devices.

AbandonedCart, 2021-11-09:
Shame it won't work as an asset for the bandwidth savings, but oh well.

AbandonedCart, 2021-11-09:
https://github.com/HiddenRamblings/TagMo/commit/4a0f39e70f56ef86a40f1ff9969f9a50663e353b should be the best of both

x7l7j8cc, 2021-11-09:
Sorry but nope...

Error again.
[image]

AbandonedCart, 2021-11-10:
I figured out the issue. GitHub Actions wasn't pulling the submodule. Newest build should fix it.