HiddenRamblings/TagMo #340: Scan, Read, Write screen

emmyexe, 2021-11-09:
Scan, Read, Write screen no longer appearing. when selecting an amiibo to write a card with it only shows a dimmed out background but the the menu does not appear

AbandonedCart, 2021-11-09:
What version and can you post the log?

emmyexe, 2021-11-09:
> What version and can you post the log?

version 2.7, 2.7.5, and the current latest build. i couldn't create a log cause the 3 dot menu refuses to open

AbandonedCart, 2021-11-09:
Interesting. What phone is it?

emmyexe, 2021-11-09:
> Interesting. What phone is it?

LG Volt
model number: LGLS740
Android version: 4.4.2

it was the cheapest phone i could obtain with NFC compatibility

Dazedfox, 2021-11-11:
I am also experiencing this issue. Even when in the settings and trying to change the option to "load photos" it just dims/grays out the screen like it wants to open something but then nothing ever happens. Unable to click on individual files or click on the three dot menu. Very frustrating :(

AbandonedCart, 2021-11-11:
No log?

Dazedfox, 2021-11-11:
I don't think I can pull a log without being able to open the three dot menu, right?

AbandonedCart, 2021-11-11:
Well, you can also pull one with adb and connecting the phone to a computer, but I figured that was probably too much to ask.

Dazedfox, 2021-11-11:
Sorry if this is not what you're looking for. I have never had to do this before. Hopefully this is the log you need!

11-10 23:58:35.104 E/dalvikvm(20293): Could not find class 'android.app.NotificationChannel', referenced from method com.pluscubed.logcat.LogcatRecordingService.handleCommand
11-10 23:58:35.104 W/dalvikvm(20293): VFY: unable to resolve new-instance 51 (Landroid/app/NotificationChannel;) in Lcom/pluscubed/logcat/LogcatRecordingService;
11-10 23:58:35.144 I/InputMethodManagerService(  744): Disable input method client, pid=20293
11-10 23:58:35.144 I/InputMethodManagerService(  744): Enable input method client, pid=1582
11-10 23:58:35.154 D/PMS     (  744): acquireWL(43c5c818): PARTIAL_WAKE_LOCK  ActivityManager-Launch 0x1 744 1000
11-10 23:58:35.164 D/AppWidgetHostView( 1582): updateAppWidget mInfo = AppWidgetProviderInfo(provider=ComponentInfo{com.pluscubed.matlog/com.pluscubed.logcat.RecordingWidgetProvider})
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve direct method 237: Landroid/app/Notification$Builder;.<init> (Landroid/content/Context;Ljava/lang/String;)V
11-10 23:58:35.164 I/RemoteViews( 1582): com.pluscubed.matlog (false,0)
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve virtual method 265: Landroid/app/Notification$Builder;.setLocalOnly (Z)Landroid/app/Notification$Builder;
11-10 23:58:35.164 I/RemoteViews.Performance( 1582): com.pluscubed.matlog 2 1 3
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve virtual method 245: Landroid/app/Notification$Builder;.setCategory (Ljava/lang/String;)Landroid/app/Notification$Builder;
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve virtual method 272: Landroid/app/Notification$Builder;.setRemoteInputHistory ([Ljava/lang/CharSequence;)Landroid/app/Notification$Builder;
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve virtual method 244: Landroid/app/Notification$Builder;.setBadgeIconType (I)Landroid/app/Notification$Builder;
11-10 23:58:35.164 E/dalvikvm(20293): Could not find class 'android.app.Notification$Action$Builder', referenced from method androidx.core.app.NotificationCompatBuilder.addAction
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve new-instance 46 (Landroid/app/Notification$Action$Builder;) in Landroidx/core/app/NotificationCompatBuilder;
11-10 23:58:35.164 W/dalvikvm(20293): VFY: unable to resolve instance field 14
11-10 23:58:35.164 I/System  (20293): exec(logcat -v time @ com.pluscubed.logcat.helper.RuntimeHelper.exec:43)
11-10 23:58:35.174 W/dalvikvm(20293): VFY: unable to resolve virtual method 288: Landroid/app/Notification;.getGroup ()Ljava/lang/String;
11-10 23:58:35.174 W/dalvikvm(20293): VFY: unable to resolve instance field 14
11-10 23:58:35.174 W/dalvikvm(20293): VFY: unable to resolve virtual method 288: Landroid/app/Notification;.getGroup ()Ljava/lang/String;
11-10 23:58:35.174 W/dalvikvm(20293): VFY: unable to resolve virtual method 288: Landroid/app/Notification;.getGroup ()Ljava/lang/String;
11-10 23:58:35.204 I/InputMethodManagerService(  744): Disable input method client, pid=1582
11-10 23:58:35.204 I/InputMethodManagerService(  744): Enable input method client, pid=20293
11-10 23:58:35.204 I/InputMethodManagerService(  744): Disable input method client, pid=20293
11-10 23:58:35.204 I/InputMethodManagerService(  744): Enable input method client, pid=1582
11-10 23:58:35.214 I/RemoteViews( 1064): com.pluscubed.matlog (false,0)
11-10 23:58:35.224 D/OnDemandProgressBar( 1064): release indeterminate drawable android.widget.OnDemandProgressBar{42971b90 G.ED.... ......I. 0,0-0,0 #102000d android:id/progress}
11-10 23:58:35.234 I/RemoteViews.Performance( 1064): com.pluscubed.matlog 1 18 9 11
11-10 23:58:35.244 I/FeedHostManager( 1582): [onResume]
11-10 23:58:35.244 I/FeedProviderManager( 1582): onResume
11-10 23:58:35.244 I/SocialFeedProvider( 1582): +onResume
11-10 23:58:35.244 I/SocialFeedProvider( 1582): updateAccounts - Accounts is no change
11-10 23:58:35.244 I/SocialFeedProvider( 1582): -onResume
11-10 23:58:35.254 D/PMS     (  744): releaseWL(43c5c818): PARTIAL_WAKE_LOCK  ActivityManager-Launch 0x1
11-10 23:58:35.404 W/NetworkScheduler( 1200): Error inserting preferred_charging_state=1 service_kind=0 persistence_level=1 tag=Measurement.PackageMeasurementTaskService.UPLOAD_TASK_TAG preferred_network_type=1 task_type=0 runtime=1636610315407 required_idleness_state=0 period=5981000 retry_strategy={"initial_backoff_seconds":{"30":0},"retry_policy":{"0":0},"maximum_backoff_seconds":{"3600":0}} target_class=com.google.android.gms.measurement.PackageMeasurementTaskService source=16 required_network_type=0 requires_charging=0 last_runtime=0 user_id=0 target_package=com.google.android.gms source_version=211213000 flex_time=2990000 job_id=-1, error message: columns tag, target_class, target_package, user_id are not unique (code 19) [CONTEXT service_id=218 ]
11-10 23:58:37.046 D/WifiNative-wlan0(  744): doString: SIGNAL_POLL
11-10 23:58:37.046 W/WifiHW  (  744): QCOM Debug wifi_send_command "IFNAME=wlan0 SIGNAL_POLL"
11-10 23:58:37.056 D/wpa_supplicant( 1059): nl80211: survey data missing!
11-10 23:58:37.056 I/wpa_supplicant( 1059): environment dirty rate=0 [0][0][0]
11-10 23:58:37.056 D/WifiStateMachine(  744): fetchRssiAndLinkSpeedNative RSSI = -59 abnormalRssiCnt = 0 newLinkSpeed = 52
11-10 23:58:37.056 D/WifiStateMachine(  744): fetchRssiAndLinkSpeedNative mLinkspeedCount = 1, mLinkspeedSum: 52
[... cut ...]
11-10 23:58:59.580 D/lights  (  744): [LedInfo] write attr ok name=/sys/class/leds/green/brightness string=0,len=1
11-10 23:58:59.580 D/lights  (  744): [LedInfo] write attr ok name=/sys/class/leds/green/blink string=0,len=1
11-10 23:58:59.580 E/AudioStreamOutALSA(  251): PCM_Write set_amp_mode,1
11-10 23:58:59.580 W/HtcAudFlgTh(  251): ThreadLoop_wrtie blocking by HAL 38730352ns, the mixerBufferTime is 10666666
11-10 23:58:59.580 D/lights  (  744): [LedInfo] write attr ok name=/sys/class/leds/green/off_timer string=0 0,len=3

AbandonedCart, 2021-11-11:
Nothing looks out of the ordinary at first glance, but it seems the one thing everyone with the issue shares is the "ext_sd" designation for the sdcard.

AbandonedCart, 2021-11-11:
If https://github.com/HiddenRamblings/TagMo/commit/d82fdb251656b9abe795b16a50bd3a683639c145 doesn't fix it, there is an alternative we can try. Hopefully that does it, though.

Dazedfox, 2021-11-11:
I tried [d82fdb2](https://github.com/HiddenRamblings/TagMo/commit/d82fdb251656b9abe795b16a50bd3a683639c145) but it did not seem to fix it at all, unfortunately :(

AbandonedCart, 2021-11-11:
What Android version are you on?

AbandonedCart, 2021-11-11:
https://github.com/HiddenRamblings/TagMo/commit/bbb098d29760d1c9faa99af5098a4e0b68271c58 includes some additional changes. There was a flaw in Android that the external storage was both being assigned different path names, depending on the manufacturer, and a locked permission that never made it into the API. This left developers of third-party apps with almost no access to the sdcard without some "colorful" tricks.

TL;DR You are a victim of poor standardization from earlier Android versions.

emmyexe, 2021-11-11:
would that explain why older TagMo versions work? so far, version 2.4.1 is the only work one that operates just fine for me.

AbandonedCart, 2021-11-11:
2.6+ removed support for Android 4.X entirely. One of the first things I did when I started was to restore it in 2.7.4

AbandonedCart, 2021-11-13:
Finally got an emulator running and found an issue. Fix will be up soon.

closed 2021-11-13 by AbandonedCart