HiddenRamblings/TagMo #342: Issue with Samsung S21 Ultra

lisatheshorty, 2021-11-09:
I have a Samsung S21 Ultra, but, when I try to write a tag, it just says "Error while writing main data (Stage 1) android.nfc. TagLostException: Tag was lost", as soon as I try to write the tag files. However, on my Samsung S21, it works absolutely fine with no issues. Both are brand new phones, bought yesterday. Not sure if it's a hardware or software issue. 

Here's what I've tried to resolve with the S21 Ultra:
Restarting the phone.
Closing down all apps and restarting only Tagmo
Uninstalling and reinstalling the latest version of Tagmo
Making sure my firmware is up-to-date
Trying without phone case
Trying with phone case

The app claims the data has been written to the tag as it shows when scanned in the app, however, when using elsewhere, the data doesn't exist. Can someone advise? Thanks

AbandonedCart, 2021-11-09:
Capture a log after it happens and post it here. Thanks

lisatheshorty, 2021-11-09:
Where do I get the log from?

AbandonedCart, 2021-11-09:
https://github.com/HiddenRamblings/TagMo/wiki/Reporting-Issues

lisatheshorty, 2021-11-09:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7507577/tagmo_logcat.txt)

AbandonedCart, 2021-11-10:
Looks like a lot of tag verification errors (bin files) and a key mismatch (usually ignored if it still loads files), but it's also performing half a write before breaking, which means it isn't failing to start and may legitimately be losing connection with the tag during a write.

lisatheshorty, 2021-11-10:
Interesting as it doesn't do it on my S21, only my ultra. I'm guessing it could be an issue with the NFC antenna?Sent from my Galaxy

AbandonedCart, 2021-11-10:
Do you have a thick case on the ultra or possibly a popsocket?

lisatheshorty, 2021-11-10:
No, neither. I have tried without the case as we and the same result occurred. Some of the .bin files wrote okay but mostly they're failing.Sent from my Galaxy

AbandonedCart, 2021-11-10:
Ill have to borrow a 21 ultra and see if I can figure out what's happening. The logs don't show anything unusual and nfc errors aren't entirely detailed.

lisatheshorty, 2021-11-10:
Oh okay, not a problem. Thank you for responding either way. Hopefully it's something simple or maybe an error on my side.Sent from my Galaxy

pastelrotx, 2021-11-10:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7508949/tagmo_logcat.txt)
I am having very similar issues on my Galaxy s10+ with tag lost errors. I was able to make an amiibo last night that worked just fine so I'm not sure what is going on. I also tried scanning the chips in Tagmo that got the tag lost error, and they appear as written. But when scanned in Animal Crossing they don't work. I also tried taking my phone case off and making the connection to the antenna as clear as possible.

AbandonedCart, 2021-11-10:
```
11-09 16:45:01.210 22912 25204 D Exception: java.lang.Exception: Invalid tag file. Tag must start with a 0x04.
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.nfctech.TagReader.validateTag(TagReader.java:39)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.nfctech.TagReader.getVerifiedData(TagReader.java:91)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:305)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.amiibo.AmiiboManager.listAmiibos(AmiiboManager.java:301)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.BrowserActivity.loadAmiiboFilesTask(BrowserActivity.java:964)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.BrowserActivity_.access$1001(BrowserActivity_.java:34)
11-09 16:45:01.210 22912 25204 D Exception: 	at com.hiddenramblings.tagmo.BrowserActivity_$13.execute(BrowserActivity_.java:405)
11-09 16:45:01.210 22912 25204 D Exception: 	at org.androidannotations.api.BackgroundExecutor$Task.run(BackgroundExecutor.java:400)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
11-09 16:45:01.210 22912 25204 D Exception: 	at java.lang.Thread.run(Thread.java:923)
```

That is because they are faulty bin files

pastelrotx, 2021-11-10:
They are in the same parent folder, should the keys be in a different, separate location?

Caelowo, 2021-11-10:
Hi, I also have a Samsung S21 ultra with the same issue! Can provide log too if it would be any help.

AbandonedCart, 2021-11-10:
Did some testing and it looks like the touch point is really sensitive. Staying toward the upper half of the phone worked, but any lower and it would lose it.

lisatheshorty, 2021-11-10:
Oh okay. I've managed to get it working a few times by moving the tag around until it triggers but it's very picky hahaSent from my Galaxy

AbandonedCart, 2021-11-10:
I'll try to increase the timeout a bit and see if it helps

lisatheshorty, 2021-11-11:
Okay, will you let us know when you've sorted it? I saw there was an update this morning but it's not fixed the issue so assumed it wasn't done yet :)

AbandonedCart, 2021-11-25:
Any update on this? Still an issue?

closed 2021-11-26 by AbandonedCart