HiddenRamblings/TagMo #344: Write issues

WolfWyrm, 2021-11-10:
I'm receiving a lot of transceive failed and tag lost errors. Currently on pixel 4a, double-checked the location of the nfc chip to make sure I was straight-on while writing. Using the latest release.

I managed to get one tag to write with 2.6.1 without error, I've tried about 10 others in both the latest release and 2.6.1 with no success

When scanned with the app, the tags show the correct amiibo and seem to allow overwrites, though when saved or scanned in-game they're showing as invalid. Could these tags be defective?

Here's my logcat - 

Google` Pixel 4a
Android SDK 31 (12)
TagMo Version 2.7.6

--------- beginning of main
11-10 17:39:01.359 15692 15709 D OpenGLRenderer: endAllActiveAnimators on 0xb400007368a5b870 (RippleDrawable) with handle 0xb40000727884b820
11-10 17:39:02.464 15692 16755 D TagWriter: Power Tag Confirmed: false
11-10 17:39:02.472 15692 16755 D TagWriter: 04810D0001173D0328480FE0F110FFEEA5000000493C19508E9E00336CCDFBEE4F3E7BCF3FAE6AC2D8437206E64E7543BD99AC8B4057829256E2331B80D213383DC5FA6A7DF69653E2F880AFB4B47B2F8F09E79602D9000101C8050233383932AD09A3CE8858E0BF4E37D8137B72939B7C05874A01620625795B997AA085B200271F36D1A7B2DFE4DBC889F4C555CE45A9D62A84FB3A90FE6CE1E0CC2B81483854CFF8E50469DB441F6E2B96EDCFBAD2A9F41B3AE983B95C0C8E3673B0235167D76169488A56DC8837E91CE6FE1923D57FE066CDEEE5CF09E76D11AA1B1AFF9DE4B47A567A038EB86DF4B00FD86D5070CA514967BDA40BA72C9A8F2E0BAED9CF7B3396C6E23D4291CFFF6B0D5C8AEF136DC6055E4F53B3F96A21E2424F8CA4A54026DDBF90C09BAFB914CD7218F68D1FCA1080C07CB77B74A23DCB5C626BED92299EF8308312676A10D6EE18F8CCAF8D568A8AFB5642946A09482BA7BDAA10A1DBE27C90C4F941EF9BAF6373A887147C1F38CA9CB17F79F3FBE3C237F9E57DCD2AE5C4BF6B4785209C585BE341799562B745EA4650B247FDF033EEBE0962BBC3B4C93DD16DDE91D7144D2ECEE510ADBC19FDE47F8D58FBADE76AC9FC3D1F7005A4D62B39CE148248C3527091561F5A4B6D8483A586A7F184AE156A019C71A9FE6131CE324CD06EE919FC304DCCEE51AEA2905BA340B6D84797854A059D56FF46B6B33CDCC026E9D601000FBD000000045F000000
11-10 17:39:02.480 15692 16755 E TagWriter: Validation successful
11-10 17:39:02.484 15692 16755 D TagWriter: 28480000F110FFEEA5000000493C1950
11-10 17:39:02.484 15692 16755 D TagWriter: Validation successful
11-10 17:39:02.491 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.497 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.504 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.511 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.517 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.524 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.530 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.537 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.543 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.550 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.556 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.563 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.570 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.576 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.583 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.589 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.595 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.602 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.609 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.615 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.622 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.628 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.635 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.641 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.648 15692 16755 D TagWriter: Wrote to page
11-10 17:39:02.664 15692 16755 E Exception: java.lang.Exception: Error while writing main data (stage 1)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:111)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:337)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity_.access$401(NfcActivity_.java:31)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity_$5.execute(NfcActivity_.java:157)
11-10 17:39:02.664 15692 16755 E Exception: 	at org.androidannotations.api.BackgroundExecutor$Task.run(BackgroundExecutor.java:400)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
11-10 17:39:02.664 15692 16755 E Exception: 	at java.lang.Thread.run(Thread.java:920)
11-10 17:39:02.664 15692 16755 E Exception: Caused by: android.nfc.TagLostException: Tag was lost.
11-10 17:39:02.664 15692 16755 E Exception: 	at android.nfc.TransceiveResult.getResponseOrThrow(TransceiveResult.java:48)
11-10 17:39:02.664 15692 16755 E Exception: 	at android.nfc.tech.BasicTagTechnology.transceive(BasicTagTechnology.java:154)
11-10 17:39:02.664 15692 16755 E Exception: 	at android.nfc.tech.MifareUltralight.writePage(MifareUltralight.java:193)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.writePage(NTAG215.java:83)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writePages(TagWriter.java:38)
11-10 17:39:02.664 15692 16755 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:108)
11-10 17:39:02.664 15692 16755 E Exception: 	... 10 more
11-10 17:39:05.601 15692 15709 D OpenGLRenderer: endAllActiveAnimators on 0xb400007368a5b870 (RippleDrawable) with handle 0xb40000727883f3d0
11-10 17:39:15.607 15692 15744 D TagWriter: Power Tag Confirmed: false
11-10 17:39:15.616 15692 15744 D TagWriter: 04810D0001173D0328480FE0F110FFEEA5000000493C19508E9E00336CCDFBEE4F3E7BCF3FAE6AC2D8437206E64E7543BD99AC8B4057829256E2331B80D213383DC5FA6A7DF69653E2F880AFB4B47B2F8F09E79602D9000101C8050233383932AD09A3CE8858E0BF4E37D8137B72939B7C05874A01620625795B997AA085B200271F36D1A7B2DFE4DBC889F4C555CE45A9D62A84FB3A90FE6CE1E0CC2B81483854CFF8E50469DB441F6E2B96EDCFBAD2A9F41B3AE983B95C0C8E3673B0235167D76169488A56DC8837E91CE6FE1923D57FE066CDEEE5CF09E76D11AA1B1AFF9DE4B47A567A038EB86DF4B00FD86D5070CA514967BDA40BA72C9A8F2E0BAED9CF7B3396C6E23D4291CFFF6B0D5C8AEF136DC6055E4F53B3F96A21E2424F8CA4A54026DDBF90C09BAFB914CD7218F68D1FCA1080C07CB77B74A23DCB5C626BED92299EF8308312676A10D6EE18F8CCAF8D568A8AFB5642946A09482BA7BDAA10A1DBE27C90C4F941EF9BAF6373A887147C1F38CA9CB17F79F3FBE3C237F9E57DCD2AE5C4BF6B4785209C585BE341799562B745EA4650B247FDF033EEBE0962BBC3B4C93DD16DDE91D7144D2ECEE510ADBC19FDE47F8D58FBADE76AC9FC3D1F7005A4D62B39CE148248C3527091561F5A4B6D8483A586A7F184AE156A019C71A9FE6131CE324CD06EE919FC304DCCEE51AEA2905BA340B6D84797854A059D56FF46B6B33CDCC026E9D601000FBD000000045F000000
11-10 17:39:15.633 15692 15744 E TagWriter: Validation successful
11-10 17:39:15.638 15692 15744 D TagWriter: 28480000F110FFEEA5000000493C1950
11-10 17:39:15.638 15692 15744 D TagWriter: Validation successful
[... 425 lines cut ...]
11-10 17:43:38.196 15692 15823 D TagWriter: Wrote to page
11-10 17:43:38.203 15692 15823 D TagWriter: Wrote to page
11-10 17:43:38.209 15692 15823 D TagWriter: Wrote to page
11-10 17:43:38.216 15692 15823 D TagWriter: Wrote to page
11-10 17:43:38.252 15692 15823 E Exception: java.lang.Exception: Error while writing main data (stage 1)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:111)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity.onTagDiscovered(NfcActivity.java:337)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity_.access$401(NfcActivity_.java:31)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.NfcActivity_$5.execute(NfcActivity_.java:157)
11-10 17:43:38.252 15692 15823 E Exception: 	at org.androidannotations.api.BackgroundExecutor$Task.run(BackgroundExecutor.java:400)
[... cut ...]
11-10 17:43:38.252 15692 15823 E Exception: 	at android.nfc.tech.MifareUltralight.writePage(MifareUltralight.java:193)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.nfctech.NTAG215.writePage(NTAG215.java:83)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writePages(TagWriter.java:38)
11-10 17:43:38.252 15692 15823 E Exception: 	at com.hiddenramblings.tagmo.nfctech.TagWriter.writeToTagAuto(TagWriter.java:108)
11-10 17:43:38.252 15692 15823 E Exception: 	... 10 more
`

WolfWyrm, 2021-11-10:
Update: Uninstalled latest version and installed 2.6.1. Seems SUPER picky, but I've gotten two more cards to write successfully with repeated attempts, usually 10+ before it writes correctly. Going to re-install latest version and attempt this method

closed 2021-11-10 by WolfWyrm

AbandonedCart, 2021-11-10:
Downgrades don't improve things here. That trend probably started back when only one new build existed and the only person still recommending it has made numerous bad recommendations.