HiddenRamblings/TagMo #345: Crash when scanning N2 Elite

burndive, 2021-11-11:
Logcat was too big to copy, so uploaded here: https://www.dropbox.com/s/muwpk06et33u8yq/tagmo_logcat.txt?dl=0

Crashes whenever N2 Elite is scanned; normal amiibo does not cause it to crash.

burndive, 2021-11-11:
More data: 
This seems to be associated with the bin files from Wumiibo.  After I did a "Build Wumiibo" and enabled "Show Downloads", this crash happens.
If I Uncheck "Show Downloads", the crash does not happen.
If I move the "Wumiibo (Decrypted)" folder to my Amiibo folder, the crash happens again regardless of "Show Downloads".

AbandonedCart, 2021-11-11:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/d7437e4723d75acf0c25cdb9f87ca23c528d9fe4 before the issue was posted.

closed 2021-11-11 by AbandonedCart

burndive, 2021-11-11:
Link to logcat from crash using latest version (504343b): https://www.dropbox.com/s/xlgpme4gqw1slal/tagmo_logcat%20%281%29.txt?dl=0

burndive, 2021-11-11:
More data: This may be related to the sheer number of bin files present.
When I include the Download files, the are 1909 total bin files.
IfI switch to a different folder with much fewer bin files (843 including the Download files), then I can scan the N2 with Download enabled.

burndive, 2021-11-11:
Experimenting in a folder with only Wumiibo files (multiple copies of the same folder), it seems to crash when scanning the N2 Elite if there are more than 1681 bin files present.