HiddenRamblings/TagMo #346: Wumiibo displayed even if Downloads folder is excluded

ishtangli, 2021-11-11:
Tested the wumiibo function. Tried to exclude the downloads folder but they were still included in the list

AbandonedCart, 2021-11-11:
Update.

ishtangli, 2021-11-11:
They are in different folders. My normal Amiibos are in /Amiibo while the wumiibo are in /Downloads/wumiibo. This results in duplicates

[image]

AbandonedCart, 2021-11-11:
Yes, but this was fixed.

ishtangli, 2021-11-11:
I just updated and still seeing duplicates. Anyway, i think it will be fixed in future release. Thanks for the hard work and the amazing app.

AbandonedCart, 2021-11-11:
I still haven't been able to duplicate this, even on a fresh install.

AbandonedCart, 2021-11-11:
Do you have an Amiibo (Decrypted) folder in your usual one? If you tried any of the really early tests, the folder was there. It can be safely deleted.

ishtangli, 2021-11-11:
Its fixed on the latest release.

closed 2021-11-11 by AbandonedCart

AbandonedCart, 2021-11-12:
I don't know what fixed it, but thanks for the update. I'll keep an eye on it so it doesn't come back.

ishtangli, 2021-11-12:
Thanks for the great work!