HiddenRamblings/TagMo #347: Issue with tags, says they're blank but fails to decrypt tag data

sir-blorps, 2021-11-12:
Hello, made an account to try to get help.

I have the right tags (NTAG215), and I've done every troubleshooting step I can think of.:
1) uninstalled and reinstalled tagmo
2) ensured files were in an accessible place (in a seperate folder NOT in the downloads folder)
3) rebooted phone
4) checked to see if tags were valid through TagWriter app (they are blank, valid signature, writable)
5) reverted back to an older build (2.6.1)
6) updated to most recent build, released a few hours ago
7) scanned actual amiibo cards to ensure it's not my phone\
8) got the most recent/updated unfixed-info and locked-secret I could find
9) redownloaded files from my computer and transferred to my phone in case that the phone download was weird

I'm wracking my brain here, obviously the tags are valid and stuff but it's like the app itself is just not liking them. That or I'm majorly dumb.

Any help would be appreciated!

AbandonedCart, 2021-11-12:
What data are you trying to decrypt if the tags are blank?

sir-blorps, 2021-11-12:
Yeah that's the weird thing, I hit write to tag and it says failed to decrypt. :/ Very confused.

AbandonedCart, 2021-11-12:
You have bad bin files then. I didn't see where it said you were writing. It verifies the files first

sir-blorps, 2021-11-12:
Wait it's user error!
I didn't realize I had to tap which amiibo i wanted to use first and hit write through that, rather than using the pink button menu.
I am so sorry I am dumb

closed 2021-11-12 by AbandonedCart