HiddenRamblings/TagMo #350: Serial randomizer leading to invalid bins

Havael, 2021-11-14:
Self explanatory title. The randomizer doesn't seem to make valid serials. So much so that they don't even appear in the amiibo search browser.

AbandonedCart, 2021-11-14:
Awesome.

Havael, 2021-11-14:
...so any ideas or...

Havael, 2021-11-14:
I didn't see anyone talking about it. so maybe it's worth looking.

AbandonedCart, 2021-11-14:
Maybe.

AbandonedCart, 2021-11-14:
Are there any specific details here? Retail or decrypted? Coins, Power Tags, N2? Anywhere to start?

Havael, 2021-11-15:
Decrypted n2

AbandonedCart, 2021-11-15:
I had a feeling. I'll look into it.

burndive, 2021-11-22:
I tried testing this: I wrote the original bin to the bank, then Edit Properties -> Generate Random -> Save (icon) -> Write

Scanning this amiibo fails in BOTW and Smash, and I'm the Switch settings menu. I did this for 10 banks, and all of them were invalid.

I prepared to try it with a real tag as well:
Select bin file -> Edit Properties -> Generate Random -> Save (icon) -> Save

The automatic filename ended with "(invalid)", so I decided not to waste a physical tag.

Havael, 2021-11-22:
> I tried testing this (using a retail bin file): I wrote the original bin to the N2 bank, then Edit Properties -> Generate Random -> Save (icon) -> Write
> 
> Scanning this amiibo fails in BOTW and Smash, and in the Switch settings menu. I did this for 10 banks, and all of them were invalid.
> 
> I prepared to try it with a real tag as well: Select bin file -> Edit Properties -> Generate Random -> Save (icon) -> Save
> 
> The automatic filename ended with "(invalid)", so I decided not to waste a physical tag.

Seems like you replicated the issue. Half way there

AbandonedCart, 2021-11-22:
> Seems like you replicated the issue. Half way there

Halfway to what, though?

burndive, 2021-11-22:
Here's the current TagMo implementation of serial randomization:
`    private static byte[] generateRandomUID() {
        byte[] uid = new byte[9];
        Random Random = new Random();
        Random.nextBytes(uid);

        uid[3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        uid[8] = (byte) (uid[3] ^ uid[4] ^ uid[5] ^ uid[6]);

        return uid;
    }

    @Click(R.id.generate_serial)
    void onGenerateSerialClick() {
        txtSerialNumber.setText(TagUtils.bytesToHex(generateRandomUID()));
    }`

Here's Amiibo_SN_Changer_V2 (written in python: https://gbatemp.net/threads/release-amiibo-bin-serial-changer.464702/): 
`   uid0 = random.randint(0,255)
    uid1 = random.randint(0,255)
    uid2 = random.randint(0,255)
    bcc0 = uid0 ^ uid1 ^ uid2
    uid3 = random.randint(0,255)
    uid4 = random.randint(0,255)
    uid5 = random.randint(0,255)
    uid6 = random.randint(0,255)
    bcc1 = uid3 ^ uid4 ^ uid5 ^uid6

    print("Using new serial: %02X%02X%02X%02X%02X%02X%02X%02X%02X" % (uid0, uid1, uid2, bcc0, uid3, uid4, uid5, uid6, bcc1))
    print("\nDecrypting %s\n" % bin_file)

    #decrypting to decrypt.bin
    print("amiitool.exe -d -k key_retail.bin -i '" + bin_file + "' -o decrypt.bin\n")
    os.system("amiitool.exe -d -k key_retail.bin -i '" + bin_file + "' -o decrypt.bin")

    try:
        with open('decrypt.bin', 'r+b') as fh:
            print("File decrypted\n")
            print("Injecting new serial\n")
            fh.seek(0, 0)
            fh.write(bytes([bcc1]))
            fh.seek(0x1D4, 0)
            fh.write(bytes([bcc0, uid0, uid1, uid2, uid3, uid4, uid5, uid6]))
            fh.close()
    except:
        print('\nAborting!\n')
        raise SystemExit(0)
`
I used Amiibo_SN_Changer)_V2 with N2 Elite to farm amiibo items in BOTW several years ago, so I know it works.

The main difference that I notice:
- TagMo is including 0x88 in the checksum for the first three bytes, and the Python code is not:
`bcc0 = uid0 ^ uid1 ^ uid2`
`uid[3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);`

burndive, 2021-11-22:
The other difference I noticed was the ninth byte being written to the 0 position, but that gets handled by setUID:
```
    public void setUID(byte[] value) throws NumberFormatException {
        if (value.length != UID_LENGTH)
            throw new NumberFormatException(TagMo.getStringRes(R.string.invalid_uid_length));

        tagData.position(UID_OFFSET);
        tagData.put(value, 0x0, UID_LENGTH - 1);
        tagData.put(0x0, value[8]);
    }
```

AbandonedCart, 2021-11-22:
Still can't reproduce the issue, but ill assume it's just the specific games.

AbandonedCart, 2021-11-23:
@Havael Can you try the latest release? I combined the code for this and the code to generate bin files, which seemed to be working.

Havael, 2021-11-23:
> Halfway to what, though? I still can't seem to reproduce the issue.

Just saying that this was the exact same thing that happened to me

Havael, 2021-11-23:
Newest version just crashes when I touch the "randomize serial" button

AbandonedCart, 2021-11-23:
Well that's... not an improvement.

Havael, 2021-11-23:
Shucks

AbandonedCart, 2021-11-23:
So the error it was throwing had nothing to do with the actual problem (that is now fixed). Forgot to set a length for the uid, but for some reason that caused the settings to think it was broken and crash the entire app.

burndive, 2021-11-23:
It no longer crashes like yesterday, but the bins still aren't recognized by the Switch or games.

When I have time, I'll try running the Amiibo Serial Changer V2 program and confirm that it works with that implementation. (And perhaps compare the results in a hex editor).

burndive, 2021-11-23:
With the latest build (0478e4b), after pressing "Generate Random Serial" on a bin file (or even if I edit something else, or if I edit nothing), hitting the "Save" icon at the top of the screen does nothing.  So there's no way to do anything with the results.

Normally hitting the Save icon would return me to the previous screen where I could click "SAVE" or "WRITE".

I can't test it with the N2 Elite interface, because when I scan it, it gives me an error: "NTAG215 required, found Unverified Tag!"

burndive, 2021-11-23:
Apparently my advice to get rid of the 0x88 in the checksum was wrong:
It looks like the serial changer was producing invalid bin files, but until the Switch's 5.0.0 firmware update the invalid bin files were being accepted. 
Reference:
https://gbatemp.net/threads/release-amiibo-bin-serial-changer.464702/page-8#post-7885408

burndive, 2021-11-24:
Another point of interest: TagMo recognizes the original amiibo file as valid, but it doesn't even show any of the serial-changed copies I've made in its browser.

When I select "Backup" on an N2 bank with an amiibo with a randomized serial number, it has "(Invalid)" in the name, just like saving a randomized serial from a file.

This means TagMo's validation is more strict than the Nintendo Switch's validation.  Nothing wrong with that, as long as TagMo is able to create amiibo with modified serial numbers that pass its own validation test.

For example, the first two digits being weeks of the year, mean that they must combine to a number 53 or less. (Not sure how they're encoded--are they characters representing the digits or binary numbers?)

AbandonedCart, 2021-11-24:
There are only 52 weeks in a year. None of that actually matters, though.

burndive, 2021-11-24:
Okay, I think I've figured it out.  That reddit post about serial numbers I was referencing is actually referring to the serial number printed on the bottom of the physical amiibo, not the digital serial number, so there aren't any restrictions on the value ranges.

Every single serial number for a real amiibo begins with 0x04, Once I set this value to always 0x04 (and fixed another problem in the python app where it was writing bcc0 first rather than fourth), the resulting amiibo bin files pass TagMo's validateTag test.

Here's what I'm recommending:
```
    public static byte[] generateRandomUID() {
        byte[] uid = new byte[9];
        Random Random = new Random();
        Random.nextBytes(uid);

        uid[0] = (byte) 0x04;
        uid[3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        uid[8] = (byte) (uid[3] ^ uid[4] ^ uid[5] ^ uid[6]);

        return uid;
    }
```

AbandonedCart, 2021-11-24:
... so basically the original code.

burndive, 2021-11-24:
It's working, but there's still a minor display issue. When you tap "Generate Random Serial", the generated serial number that gets displayed has a random first byte, not 0x04. Though when it gets saved and written back to the N2 bank or a bin file, it has the correct value.

burndive, 2021-11-24:
I'm sorry to be the source of confusion.  The thing missing from the "original code" that I added in `generateRandomUID` was `uid[0] = (byte) 0x04;`, which I thought would be obvious, but apparently my local copy of the source was out of date.  I should have based it on the latest code in github:
(In Foomibo.java):
```
    public static byte[] generateRandomUID() {
        byte[] uid = getRandomBytes(9);
        uid[0x0] = (byte) 0x04;
        uid[0x3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        uid[0x8] = (byte) (uid[3] ^ uid[4] ^ uid[5] ^ uid[6]);
        return uid;
    }
```

burndive, 2021-11-24:
I spoke too soon about it working, despite beginning with 0x04;  I tested in BOTW, and none of them scan.

This is because the checksum (uid[0x3]) is set in generateRandomUID, but it uses the original (random) value for uid[0] to calculate it.  `uid[0x0] = (byte) 0x04;` needs to be set before `uid[0x3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);`.

AbandonedCart, 2021-11-24:
x = 7, y = 8, z = 9;
x = 7, z = 9, y = 8;
y = 8, x = 7, z = 9;
y = 8; z = 9; x = 7;
z = 9, x = 7; y = 8;
z = 9; y = 8; x = 7;

What do all of these have in common? In every instance, x = 7, y =8, and z = 9. The order you set uid[0] and uid[3] doesn't matter if they aren't written until both are set.

burndive, 2021-11-24:
I'm not sure how else to convince you:
```
# cat Test.java && javac Test.java && java Test
public class Test {
        public static void main(String[] args) {
                byte [] uid = {(byte) 0x98, (byte) 0x86, (byte) 0x54, (byte) 0x32};
                uid[3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
                uid[0] = (byte) 0x04;
                System.out.println("Checksum calculated before setting value " + uid[3]);
                uid[3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
                System.out.println("Checksum calculated after setting value " + uid[3]);
        }
}
Checksum calculated before setting value -62
Checksum calculated after setting value 94

```

AbandonedCart, 2021-11-24:
That's not the code being used, though. That is the python code converted to Java, which was thrown out after 20 failed attempts to make it work that were all "definitely working" until the next one.

AbandonedCart, 2021-11-24:
@Havael Once you can confirm it works, we can wrap this one up.

burndive, 2021-11-24:
This is how you changed generateRandomUID in 3a8510c:
```
    public static byte[] generateRandomUID() {
        byte[] uid = getRandomBytes(9);
        // uid[0x3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        // uid[0x8] = (byte) (uid[3] ^ uid[4] ^ uid[5] ^ uid[6]);
        byte bcc = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        uid[0x0] = 0x04;
        uid[0x3] = bcc;
        uid[0x8] = bcc;
        return uid;
    }
```
uid[3] and uid[8] are two separate checksums for separate sets of bytes; they are not both bcc, and bcc will be the wrong value for uid[0x0] unless you first assign uid[0x0] = 0x04 (since uid[0x0] is used to calculate uid[0x3]).
Also, the second checksum value uid[0x8] includes uid[7] and not uid[3]. (I missed this in my earlier comment).

The correct code would be:
```
        uid[0x0] = 0x04;
        uid[0x3] = (byte) (0x88 ^ uid[0] ^ uid[1] ^ uid[2]);
        uid[0x8] = (byte) (uid[4] ^ uid[5] ^ uid[6] ^ uid[7]);
```

That is: the first three bytes are xored together with 0x88 to make the fourth byte.  The next 4 bytes are xored to make the 9th (which is eventually set as byte 0 of the entire amiibo file).

This is the working python code:
```
    uid0 = 0x04
    uid1 = random.randint(0,255)
    uid2 = random.randint(0,255)
    bcc0 = 0x88 ^ uid0 ^ uid1 ^ uid2
    uid3 = random.randint(0,255)
    uid4 = random.randint(0,255)
    uid5 = random.randint(0,255)
    uid6 = random.randint(0,255)
    bcc1 = uid3 ^ uid4 ^ uid5 ^uid6
```
But keep in mind that uid4, uid5, and uid5 correspond to your uid[5], uid[6], and uid[7], since this is how they are written to the file:
```
            fh.seek(0, 0)
            fh.write(bytes([bcc1]))
            fh.seek(0x1D4, 0)
            fh.write(bytes([bcc0, uid0, uid1, uid2, uid3, uid4, uid5, uid6]))
```

AbandonedCart, 2021-11-24:
If the only version of the code that is acceptable is one in an archaic python script, stick with that.

burndive, 2021-11-24:
The python script needs to be run on a Windows computer, then you have to copy all the bin files to your phone and select them one by one.  Integrated into tagmo, all you need is one original bin file, and you can create as many randomized copies in place on an N2 or written to physical tags.

I've created my own fork and submitted a pull request.  I tested it on my Switch.  @Havael you can download it here to test: https://github.com/burndive/TagMo/actions/runs/1500918675

burndive, 2021-11-24:
@Havael The procedure for testing:
1. Write original .bin file to N2 bank
2. On that Bank -> Edit Properties (scan)
3. Generate Random Serial
4. Tap Save icon
5. WRITE (scan)

For a physical tag:
1. Select bin file
2. Edit Properties
2. Generate Random Serial
3. Tap Save Icon
4. WRITE (scan)

commit 9203a099 (2021-11-16): Fix performing full write to N2 (fixes #350)
closed 2021-11-24 by AbandonedCart