HiddenRamblings/TagMo #352: Can we use the editor smash 4 for smash ultimate?

BigBoss320, 2021-11-15:
Hello. I have a question:
Can I edit an amiibo with Smash 4 abilities to use it on Smash ultimate?
Because when I add an amiibo that I used on ultimate, it shows "Unknown" on Tagmo Editor and I can't modify it. 
If no, is it possible to add it?
Thank you, you've made a great job.

AbandonedCart, 2021-11-15:
Have you followed https://github.com/HiddenRamblings/TagMo/wiki/Instructions#editing-amiibo-properties?

BigBoss320, 2021-11-15:
Yes but there isn't Super Smash bros Ultimate. It's only for Smash Bros Wiiu/3ds

AbandonedCart, 2021-11-15:
And you tried that, but it didn't work?

BigBoss320, 2021-11-15:
I didn't tried it yet, and I can't right now.

AbandonedCart, 2021-11-15:
Alright. I guess we can leave this open until we know the current state.

BigBoss320, 2021-11-15:
Yes, sorry I have classes I can try it later at about 6pm (Paris, France hours)

burndive, 2021-11-15:
Smash Ultimate uses a different format for amiibo fighter data than Smash for Wii U/3DS.  If you have an amiibo with Wii U/3DS Smash data on it, Smash Ultimate will offer to convert it to its own format, but this breaks compatibility with previous versions.

Here's a thread on the format, where people started developing an editor:
https://gbatemp.net/threads/smash-bros-ultimate-amiibo-editor.525909/

BigBoss320, 2021-11-15:
So if I understand right, we can't edit the damage and the defense without "corrupting" the amiibo in-game?

AbandonedCart, 2021-11-23:
Looks like https://github.com/odwdinc/SSBU_Amiibo is the current status. There is a lot of stuff in there TagMo can't include, since remaining on the right side of the law means not including any assets (images, audio, dialog). I'm not entirely sure if the editor works, since the thread about it is a bit mixed. I reached out to them to see if they had any interest in having it included, so I guess we'll see. As it stands now, my python is a bit rusty and converting that is not something that will happen nearly as fast as some of the other features.

AbandonedCart, 2021-11-26:
There has been no response yet, so I have started looking through the code. Discarding everything that is part of the standard processing of a bin file, it looks like the following 4 files make up the unique functionality.

https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ui.py
https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ssbu_amiibo/amiibo_class.py
https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ssbu_amiibo/hex.py
https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ssbu_amiibo/trainingData.py

BigBoss320, 2021-11-27:
Okay, I will look at this. Thanks

AbandonedCart, 2022-01-10:
After doing a little more research, the following information appears to be the core functionality to perform edits. Everything else appears to make up the UI, hex editor, and general processing of the bin file.

https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ssbu_amiibo/trainingData.py - 
```
DatablockOffest= 0xE0
un0Offest= 0x03
un1Offest= 0x0F
un4Offest= 0x76

knowLocations = {
'un0':{},
'un1':{},
'un4':{

		0x168-DatablockOffest-un4Offest:	"Attacking???",
		0x169-DatablockOffest-un4Offest:	"Attacking???",
		0x16A-DatablockOffest-un4Offest:	"Attacking???",
		0x16B-DatablockOffest-un4Offest:	"Attacking???",

		0x181-DatablockOffest-un4Offest: 	"Taunting! its value based, so the greater the Hex Value the more the Amiibo will taunt!",
		0x171-DatablockOffest-un4Offest: 	"Grabbing! Also Value Based, but not as great as Taunting.",

		0x18A-DatablockOffest-un4Offest:	"A special Neutral",
		0x18B-DatablockOffest-un4Offest:	"A special Side",
		0x18C-DatablockOffest-un4Offest:	"A special Up",
		0x18D-DatablockOffest-un4Offest:	"A special Down",

		0x19A-DatablockOffest-un4Offest: 	"B special Neutral",
		0x19B-DatablockOffest-un4Offest:	"B special Side",
		0x19C-DatablockOffest-un4Offest:	"B special Up",
		0x19D-DatablockOffest-un4Offest:	"B special Down",
	}
}
```

https://github.com/odwdinc/SSBU_Amiibo/blob/master/src/ssbu_amiibo/amiibo_class.py

BigBoss320, 2022-01-10:
Wow thank you it can be helpful to edit the tag, thx again

AbandonedCart, 2022-01-10:
It may be a little easier to use the python package already available than trying to manually edit the files based on the code.

AbandonedCart, 2022-07-24:
Closing this in favor of future discussion for native SSBU support

closed 2022-07-24 by AbandonedCart