HiddenRamblings/TagMo #353: Failed to decrypt tag

cocconutdoggo, 2021-11-15:
So i recently bought some NFC Sticker off of Amazon and when i tried them today no matter what i tried with the keys and everything it just seems to not want to work... Ive tried scanning other NFC Cards that i bought previously (already written on unfortunatly) and they seem to work just fine... any ideas how i can fix it? and hopefully telling me i didnt waste my money on these stickers.

AbandonedCart, 2021-11-15:
Do you have a log? It's possible they are bad bin files or there is a bug, but it is impossible to tell which based on the message alone.

cocconutdoggo, 2021-11-15:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7540651/tagmo_logcat.txt)
hope this works

closed 2021-11-15 by cocconutdoggo

AbandonedCart, 2021-11-15:
You're using a really old version. There is a good chance the issue was already fixed, but we can investigate further if it wasn't.

cocconutdoggo, 2021-11-15:
I was just dumb and forgot to follow one step.