HiddenRamblings/TagMo #361: Editing amiibo properties in N2 banks errors out in recent releases

AuroraWright, 2021-11-18:
Not sure which commit/release broke this (the 2.7.5 tag appears to work properly) but "edit properties" on an N2 bank on latest shows "Failed to display tag data"

AbandonedCart, 2021-11-18:
Log?

AuroraWright, 2021-11-18:
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7564844/tagmo_logcat.txt)

AbandonedCart, 2021-11-18:
Fixed by https://github.com/HiddenRamblings/TagMo/commit/a8ce089fb64f0811f27e4aec01e7ec5bf8c002dc

Some previously unencrypted data (that didn't work with consoles anyway) may still not load properly.

AuroraWright, 2021-11-19:
I tested TagMo-c345449-signed.apk and I still experience the same issue, logcat seems to be the same as before
[tagmo_logcat (1).txt](https://github.com/HiddenRamblings/TagMo/files/7566831/tagmo_logcat.1.txt)

AbandonedCart, 2021-11-19:
Do the same slots work with a console?

AuroraWright, 2021-11-19:
Yes, and editing works properly with the 2.7.5 build

AbandonedCart, 2021-11-19:
The 2.7.5 build didn't verify anything. The errors in your log are they keys being in the same folder as the bin files. The new log didn't catch any attempt to load a bank.

AuroraWright, 2021-11-19:
I took a video of the issue, not sure how I can help more (if I can test anything for you let me know)
https://user-images.githubusercontent.com/15571693/142518474-06657ee2-d49f-468a-bd8e-195965bcd246.mp4

AbandonedCart, 2021-11-19:
I believe you that an issue exists. It's just unclear what is causing it. It seems sporadic.

AbandonedCart, 2021-11-19:
It looks like the N2 works fine with completely unencrypted data (as in 2.7.5), but editing the data requires it to be properly "signed" to work (https://github.com/HiddenRamblings/TagMo/issues/350). It is almost impossible to support both formats at the same time. If we add support for the old formatting, it breaks any attempts to make sure things work.

That said, there is a workaround. If you perform a backup of the slot, then "replace" and write the backup, it fixes it.

AbandonedCart, 2021-11-19:
I will also work on coming up with a cleaner (and easier) way to convert the data, which will hopefully happen behind the scenes without all the extra steps.

AuroraWright, 2021-11-19:
Thank you for your explanation/fix, i just tried the latest build, repaired the amiibo and it seems to still error out but the error changed:
11-19 18:58:44.449 29395 29425 W TagReader: 04FCAADA72B2488008480FE0F110FFEEA50000001A53526DA13D71C04353253268FC981446D6C4004F27520821F362CABA8D4DE0D3726BA043A9F6806C6B751033E4A1E44E45171FDE90420B4A4507A3FB4BBF6B00030102004303020C12551852D8782624765AB41752BA72DA796E4757E371BCF742739F0043C62C9DD7A9C79ED8C1DCACBF3AF18E21A1487296BD3F4CA674CD817C91B049831E2F88B05E35F532EE3D5ECC195193323A2D5107874B6AB7F9D1FE95CBDAC1BC61C6647DB52D82EA1DDF3251A06059EE00A4751633528C6D86E42F5844441572A20BB56C3EED33E8453205796BB30D9B0B1611A03D6D8CDF07849A01CF151628DD2140D5F1C1EB61B360725CFC6CBA7FA7412A618B293BB2642384177507EFC485CC0810E6239344F9BDFEC262D2420CD13558F4B5E8B731E761F313F38C7A11DC58EE5994564C9A3424DF91C0F1E21F1205AEA9CC761E10C6505EC2E7E87EE72198C761D300B57BDAE0B7B52C29B470D8A1EC22682DF61808FA6068B1D30A170C6A1A55E29C770BEE908C8CA3B44144750798EDBEF4BE54CE675DA052FB3877D7798E144D0765DEB4BD7FAEF8426BA521748BC7DD0B604CAD970DF0CA533857FB43E5C8AA363001218B6B0B3915E6369DB8C48BB3CAE7E43DE6432BCB1F8E3B457142B97359764DC2A5211E391F0086199B839A9AC358EE8555AE59F041DED35D7A74F61F45832F652C8664CF7301000F00000000045F000000244D906780800000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
11-19 18:58:44.744 29395 29395 W Exception: java.lang.Exception: Failed to decrypt tag data
11-19 18:58:44.744 29395 29395 W Exception: 	at com.hiddenramblings.tagmo.amiibo.KeyManager.decrypt(KeyManager.java:131)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.hiddenramblings.tagmo.TagDataActivity.afterViews(TagDataActivity.java:166)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.hiddenramblings.tagmo.TagDataActivity_.onViewChanged(TagDataActivity_.java:187)
11-19 18:58:44.744 29395 29395 W Exception: 	at org.androidannotations.api.view.OnViewChangedNotifier.notifyViewChanged(OnViewChangedNotifier.java:41)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.hiddenramblings.tagmo.TagDataActivity_.setContentView(TagDataActivity_.java:71)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.hiddenramblings.tagmo.TagDataActivity_.onCreate(TagDataActivity_.java:54)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.Activity.performCreate(Activity.java:7994)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.Activity.performCreate(Activity.java:7978)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1316)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3406)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3597)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2068)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.os.Handler.dispatchMessage(Handler.java:106)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.os.Looper.loop(Looper.java:223)
11-19 18:58:44.744 29395 29395 W Exception: 	at android.app.ActivityThread.main(ActivityThread.java:7668)
11-19 18:58:44.744 29395 29395 W Exception: 	at java.lang.reflect.Method.invoke(Native Method)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
11-19 18:58:44.744 29395 29395 W Exception: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954)

The interesting part is that editing properties does work just after a write to the N2 slot happens (either replacing or repairing), but as soon as you close the N2 window and "Scan NFC tag" again, it triggers the error again

AbandonedCart, 2021-11-19:
https://github.com/HiddenRamblings/TagMo/commit/4f8ab5f75cec353f2efdca3867884cbc1350aaeb should fix it for good

burndive, 2021-11-20:
I was seeing this issue with Wolf Link (with Twilight Princess heart data) and amiibo with Smash Bros 4 data.  I tested with both, and it works now.

AuroraWright, 2021-11-20:
Still seeing the same issue with (seemingly) the same logcat on d5beb6b. Tagmo is now offering to repair the amiibo automatically after the error shows up, if you do it "edit properties" works until the N2 window is closed and the N2 is scanned again (as before). Tried it with multiple amiibo
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7573899/tagmo_logcat.txt)

AbandonedCart, 2021-11-20:
What was the original file that was written to that bank? Was it a retail clone, decrypted, or "imported from somewhere else"?

AuroraWright, 2021-11-20:
Can't reproduce it anymore with latest commit, thanks for your patience heh

AbandonedCart, 2021-11-20:
Glad to hear it finally worked out. Let me know if it ever starts up again.

AuroraWright, 2021-11-20:
It seems the issue is not completely fixed after all, as if there is any data at all on the amiibo it appears to be encrypted/decrypted to garbage when displayed (?)
Steps to reproduce:
- Register a mii/nickname on a legitimate amiibo with a console
- Dump the amiibo with Tagmo, verify that the information is correctly shown using "edit properties"
- Write amiibo to N2, close/reopen N2 window, click "edit properties" on the bank
- Data is completely random, as if encrypted or decrypted to garbage

If the amiibo is empty, data is correctly displayed as being empty however.
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7575421/tagmo_logcat.txt)

AbandonedCart, 2021-11-20:
Somehow your phone is pulling the banks without encryption and passing every single check the they are encrypted and then displaying them without it. I honestly don't know how to fix that.

AuroraWright, 2021-11-20:
I see, for what it's worth I tried with both a Pixel 2 XL with an AOSP 11-based ROM and a recent Xiaomi phone and both seem to act the same way

AbandonedCart, 2021-11-20:
I'll have to see if I can make some data happen that will throw a wrench in it on my end. There has to be something going wrong somewhere, but the logs aren't catching it. I need to clean up the validation code anyway. It was becoming a mess.

AuroraWright, 2021-11-24:
Apparently it's fixed now, not sure which commit did it. I successfully edited nickname/game data directly in banks several times (even created new data from scratch in empty ones) so at this point it should be safe to close this. (I found an unrelated issue at the same time which I'll open a new issue about)

AbandonedCart, 2021-11-24:
I have a feeling trying to fix a different issue caused this one. I undid those changes, but I'm glad it worked.

closed 2021-11-24 by AbandonedCart