HiddenRamblings/TagMo #362: TagMo continuousky crashes after New Releases until Updated

Invigorater, 2021-11-18:
I have a Samsung Galaxy S8 on Android Version 9 with the latest release of TagMo listed here.

Whenever a new release of TagMo is issued, my TagMo app on my phone will constantly crash after opening until I come here to manually update my app with the latest apk file.

I am not sure if this is intentional but just wanted you to be aware.

AbandonedCart, 2021-11-18:
Log?

Invigorater, 2021-11-18:
So I'm trying to follow your reporting instructions to pull the logcat file but I can't even have the app stay open long enough to save one without it crashing. 

Is there an alternate way to generate a log file? FYI, my TagMo app is currently having this issue.

AbandonedCart, 2021-11-18:
https://github.com/HiddenRamblings/TagMo/wiki#logcat-with-instant-crash

Invigorater, 2021-11-18:
So I was finally able to beat the crash and generate a logcat and also found a crash logcat just ina different folder listed from your instructions.

[crash_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7566580/crash_logcat.txt)

[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7566581/tagmo_logcat.txt)

Invigorater, 2021-11-18:
I will wait to update my TagMo app until I hear back from you if you need further logs or tests.

closed 2021-11-18 by AbandonedCart

Invigorater, 2021-11-20:
@AbandonedCart

I did have a few crashes last night so I attached an updated crash logcat and was able to generate another logcat.

[crash_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7574558/crash_logcat.txt)

[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7574559/tagmo_logcat.txt)