HiddenRamblings/TagMo #363: Backup bin files are saved to Download folder

burndive, 2021-11-20:
I expect bin files to be saved to The TagMo Backup folder (or so least somewhere within the working directory, where they can be used).

AbandonedCart, 2021-11-20:
Enable "show downloads" and use them in place. Grab them from the downloads folder and put them where you want them to use them. Most people have a very organized collection. Throwing a random backup folder in there tends to get lost in what was otherwise a very specific layout. As proven by a few different reports of duplicates, the normal expectation was that such items were not automatically thrown into the working folder. This is all on top of the app flat out telling you where it puts the files.

closed 2021-11-20 by AbandonedCart

burndive, 2021-11-22:
Can we at least have the option in Settings to specify the folder?  

Putting them in Downloads implies that they were downloaded, which is not at all the case.  Also, the Downloads folder has a bunch of random crap in it, which is the definition of disorganization.  Why should TagMo be sifting through that, and why should I have to move a file every time I back up an amiibo?  I should be able to tell TagMo where to put scanned amiibo, and they should all show up there without making me open a file manager.