HiddenRamblings/TagMo #365: Increasing N2 bank count freezes the app

AuroraWright, 2021-11-24:
When increasing the N2 bank count, the app crashes and needs to be reopened. The change occurs successfully though. On a related note, reducing the bank count doesn't cause a crash but the N2 window doesn't get refreshed.
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7598675/tagmo_logcat.txt)

AuroraWright, 2021-11-24:
Latest commit works fine increasing but crashes decreasing
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7599135/tagmo_logcat.txt)
:

AuroraWright, 2021-11-25:
There's no more crashing now, but after decreasing it seems the wrong banks are shown:
[image]
[image]

AbandonedCart, 2021-11-25:
Lol. Android really broke adapters...

AbandonedCart, 2021-11-25:
Included in https://github.com/HiddenRamblings/TagMo/commit/3c5d76026ea200f010ab01b73aa45f8f4decefb6

closed 2021-11-25 by AbandonedCart