HiddenRamblings/TagMo #367: Serial randomizer fixed in fork

Havael, 2021-11-24:
@Havael Sorry you got steamrolled on your own issue. Please open a new one if the issue persists.

_Originally posted by @AbandonedCart in https://github.com/HiddenRamblings/TagMo/issues/350#issuecomment-978070561_

So it seems that fork was working on the given issue. But, since it's very unstable, I would like to know if that solution will be implemented. While avoiding crashing randomly after some time(or rotating the screen).

AbandonedCart, 2021-11-24:
I don't know what fork that is, but a solution has been implemented.

Havael, 2021-11-24:
> I don't know what fork that is, but a solution has been implemented.

Oh. I guess I'll try before saying the fork

Havael, 2021-11-24:
By the looks of it. It's working. But it would be nice to edit properties in the n2 banks directly. Maybe another issue

closed 2021-11-24 by Havael

AbandonedCart, 2021-11-24:
I'm not sure I understand what that means.