HiddenRamblings/TagMo #368: Tagmo incorrectly indentifying n2 as NTAG215

Havael, 2021-11-24:
Self explanatory. Not much to add

Havael, 2021-11-24:
When scanning

AbandonedCart, 2021-11-24:
It's popping up as a single tag?

Havael, 2021-11-24:
Yeah

AbandonedCart, 2021-11-24:
Make sure n2 support is enabled in settings

Havael, 2021-11-24:
> Make sure n2 support is enabled in settings

It is

AbandonedCart, 2021-11-24:
Unlock the hardware

Havael, 2021-11-24:
> Unlock the hardware

Done also

Havael, 2021-11-24:
It just started doing this

AbandonedCart, 2021-11-24:
Is your N2 going bad?

Havael, 2021-11-24:
> Is your N2 going bad?

Nope. Just tested it with the n2 elite app and, sure enough, working as usual

Havael, 2021-11-24:
I think this happened before with an older version of tagmo. But it just solved itself, so no idea what happened

AbandonedCart, 2021-11-24:
There are only 3 causes for it. N2 support disabled, locked hardware, and corrupt signature.

Havael, 2021-11-24:
Sorry. Worded the problem wrongly

Havael, 2021-11-24:
It thinks it's an unsupported format

AbandonedCart, 2021-11-24:
Oh. Scan it in the official app and change active banks. The recovery code isn't in TagMo yet.

Havael, 2021-11-24:
Oh. I'm stupid. I was trying to scan a bank I emptied

Havael, 2021-11-24:
Maybe there should be a clearer error message. But it's probably my smooth brain talking. Sorry to bother

AbandonedCart, 2021-11-24:
It's not our error message. That's the NFC hardware.

Havael, 2021-11-24:
> It's not our error message. That's the NFC hardware.

Ah I see. Maybe put a warning in the wiki, if there isn't already, so people don't make this mistake or smth

Havael, 2021-11-24:
> It's not our error message. That's the NFC hardware.
> 
> What I've tried to do is catch the ones with known solutions and offer that instead of the vague error.

Well that's better at least. Thanks

AbandonedCart, 2021-11-24:
You shouldn't be able to scan a blank bank unless you keep TagMo open to the N2 browser, swap to the N2 app and clear one, then return. As for the other cause, that should get resolved by overwriting any banks with the bad serial numbers. I wouldn't know what sort of warning to give, other than not to use old builds. I think it does say to update somewhere.

closed 2021-11-25 by AbandonedCart