HiddenRamblings/TagMo #369: Attempt to get length of null array

riley-west, 2021-11-26:
I'm getting the above message when scanning and writing tags. Not sure if it's an error or not but it comes up in big red text.

AbandonedCart, 2021-11-26:
Are you scanning empty tags and trying to write amiibo you don't have?

riley-west, 2021-11-26:
To be clear, the message shows up both when using the data validation option or the writing tag option and is successful.  But it appears during both processes.

AbandonedCart, 2021-11-26:
That wasn't what I was asking, though. I was asking if you are writing valid bin files and scanning tags with data on them.

riley-west, 2021-11-26:
Yes & yes.

AbandonedCart, 2021-11-26:
Looks like you're using an old version. The lines referenced in the log don't exist anymore.

riley-west, 2021-11-26:
It's the newest version with the self update enabled.

He's another scan of another tag done just now
[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7605769/tagmo_logcat.txt)
.
I also did a screen capture showing the message appearing.
[https://streamable.com/k4gs3q](https://streamable.com/k4gs3q)

AbandonedCart, 2021-11-26:
java.lang.NullPointerException: Attempt to invoke interface method 'void java.util.List.sort(java.util.Comparator)' on a null object reference
11-22 09:07:37.490 21572 21572 E AndroidRuntime: 	at java.util.Collections.sort(Collections.java:206)
11-22 09:07:37.490 21572 21572 E AndroidRuntime: 	at com.hiddenramblings.tagmo.adapter.BrowserAmiibosAdapter$AmiiboFilter.publishResults(BrowserAmiibosAdapter.java:245)

https://github.com/HiddenRamblings/TagMo/blob/master/app/src/main/java/com/hiddenramblings/tagmo/adapter/BrowserAmiibosAdapter.java#L245

AbandonedCart, 2021-11-26:
Power Tag or card / coin?

riley-west, 2021-11-26:
They're single use tags.  I do have N2 compatibility enabled.

I just disabled N2 in the settings and the error goes away.

AbandonedCart, 2021-11-26:
That's because the hardware will instantly fail on the check for an n2 tag. Yours doesn't. It needs a secondary check for if it doesn't.

riley-west, 2021-11-26:
Oh ok, so it's not an error.  Good to know, thanks for explaining.

closed 2021-11-26 by AbandonedCart