HiddenRamblings/TagMo #371: TagMo quits (no LogCat) when scanning N2 if too many bin files are present

burndive, 2021-12-02:
When I scan my N2, TagMo flashes a different screen, and then exits;  I would say it crashes, but there is no LogCat file in my Downloads/TagMo/LogCat folder.

I switched to a different working directory, and duplicated folders full of bin files until N2 wouldn't scan anymore.  Then I started deleting smaller groups of bin files until it would scan again.

I have 95 banks enabled on my N2 Elite.
At 1664 bin files, N2 scan causes the app to quit.  At 1663 bin files, it scans.
After increasing the N2 bank count to 100, with 1663 bin files present, it quits, but if I delete 5 bin files (down to 1658), it scans.

**So, it looks like the sum of the bin files plus enabled banks must be 1758 or less for N2 to scan without the app quitting.**

My device is a Samsung Galaxy S9+.

Invigorater, 2021-12-03:
@AbandonedCart 

I can confirm that I am having the same issue as @burndive. No crash_logcat is generated but my TagMo closes after trying to scan the N2. I'm trying to generate a tagmo_logcat but TagMo is not creating one with the following error. I am using a Samsung Galaxy S8 on Version 9 Android.

[image]

AbandonedCart, 2021-12-03:
The log issue should be resolved, but an OoM may not be something the app can address. If the phone limitations prevent more than a certain number of items, that is the number of items. 6GB RAM may seem like a lot, but Android only allows so much per application.

Invigorater, 2021-12-03:
Here is a tagmo_logcat:

[tagmo_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7652881/tagmo_logcat.txt)

AbandonedCart, 2021-12-03:
Looks like somewhere it is trying to convert an integer to a string or vice versa. That may be a different issue.

AbandonedCart, 2021-12-04:
@Invigorater The issue showing up in your logs has been addressed in https://github.com/HiddenRamblings/TagMo/commit/c68bb25cce0df02b5326d7cf3c1ad101d6cd24d7

AbandonedCart, 2021-12-04:
I'm inclined to believe it's a device issue. I just double checked and I have 1754 files and 25 banks. That's already over 1758, but I enabled all 200 banks and still couldn't crash it.

Invigorater, 2021-12-04:
It's interesting which folder I select with the down arrow as it hides more files the deeper I go to get me below the threshold that burndive mentioned.

What is this Foomiibo folder located under Downloads/TagMo/Foomiibo?

I don't recall putting it there and is where the large amount of duplicate bins are that are causing this issue. I have a completely separate folder that I use to maintain my bins.

AbandonedCart, 2021-12-04:
You can delete it if you don't want it, but it's generated by using the Foomiibo menu item.

AbandonedCart, 2021-12-05:
Some of the checks to prevent displaying folders twice were a little convoluted from attempting to make them more accurate. Actually, the easiest way to check was if the path of one is the start of the other's path, so that should make things run a little smoother.

There are also about 20 different changes to optimize and improve the bitmap handling. It's a pretty common thing that when there is what looks like an OutOfMemory error, it's likely some Bitmap(s) somewhere at the root of it.

closed 2021-12-06 by AbandonedCart