HiddenRamblings/TagMo #374: Cant Install the Version dec1a9b

Ericzion1999, 2021-12-09:
I have the Samsung a3 (2017) With Android 8 , When i try to Install the Apk , my Phone says ,,App was not installed,,[image]

AbandonedCart, 2021-12-09:
Current versions can't be installed directly over those from 2020 and earlier.

Ericzion1999, 2021-12-09:
I don't have tagmo installed before

Ericzion1999, 2021-12-09:
I have it installed Version 3cb304b now on my HTC one but it Crashes everytime If i want to Open the App 
[logcat](https://drive.google.com/file/d/1B-bv8sDX9O32w7xB9f8Fm5mInLI7gT4T/view?usp=drivesdk)

AbandonedCart, 2021-12-09:
There's no way of knowing why it won't install without running logcat during the install. That's Android stopping it. The most common reasons being installing with a different signature and not having the hardware for a required feature (NFC is the only one)

Ericzion1999, 2021-12-09:
This one ?
https://play.google.com/store/apps/details?id=com.google.android.gms

Ericzion1999, 2021-12-09:
It works now on my HTC one 0493ec5
I think IT doesnt Work on my Samsung because i have Not enough space , sorry for the issue

closed 2021-12-11 by AbandonedCart