HiddenRamblings/TagMo #375: After scanning N2 and pressing the bottom bar, it shows an error and not the active bank

riley-west, 2021-12-11:
Screenshot attached. 
[image]

AbandonedCart, 2021-12-11:
It never showed the active bank, but clicking on the active bank should display correctly. The view isn't being removed when nothing it loaded. No functionality has actually broken.

riley-west, 2021-12-11:
Just seemed like a missed opportunity to show *anything* when hitting the bottom bar.

Thanks for everything you do!

AbandonedCart, 2021-12-11:
Thanks for the idea. Displaying the active bank as the initial default is better than starting blank. It may not make sense for every time it was blank, but it definitely works there.

closed 2021-12-11 by AbandonedCart