HiddenRamblings/TagMo #377: I can't open app.

Entrotom, 2021-12-16:
Hello. Whenever I try to open app, crashes happen and I can't use this app. I tried to download all version app. The olds can be opend but tag is not recognized, and new versions are crashed. I want to attach my crash log, but there's even no log in tagmo folder. My device is SAMSUNG GALAXY 10+. I really want to use this app and  need help. Thanks.

AbandonedCart, 2021-12-16:
Without a log, there is no way of knowing why TagMo crashed.

https://forum.xda-developers.com/t/tutorial-how-to-logcat.1726238/

AbandonedCart, 2021-12-16:
Just to be sure, are you checking the "TagMo" folder in Download/TagMo or were you looking in Android/data/com.hiddenramblings.tagmo/files because the first one is where the log should write.

3afd8e76cc91d219d3549ac4a9ad703c6a55da02 adds a fallback to the second directory if writing to the first fails for some reason, but versions before that ONLY write to the Download folder.

closed 2021-12-18 by AbandonedCart