HiddenRamblings/TagMo #378: TagMo needs a guide for editing properties, and an explanation of what Foomiibo is.

mtjohnson47, 2021-12-19:
Using TagMo is largely guesswork, and I can't find the info from searching Google.

Should I set the Country Code to my Switch region, or leave it unset?  Should I change the initialized/modified dates?  Should initialized be date of manufacture?  Does Nickname matter?

I'm guessing write counter is how many times the Amiibo has been written to.  Would setting it to 0 make it seem like a brand new from factory Amiibo to the switch?  Or toggling the properties checkbox for the same effect?

Should I randomize the serial number to avoid detection by Nintendo, or would randomizing it make it easier to detect?

Should Mii Name by my Mii name or blank?

What is Foomiibo?  It seems like an option to duplicate amiibo bins so you can edit the properties without editing the original bin.  Is that correct?  This should be explained somewhere.

Lastly, the Power Tag and N2 Elite support checkbox say to see github for limitations and use, but Github doesn't show the limitations or use.

mtjohnson47, 2021-12-19:
Playing around with it more, itlooks like maybe Foomiibo generates all 826 Amiibo bin files, leaving each blank with a blank random ID?

AbandonedCart, 2021-12-19:
If you don't know what properties do, they are best left alone. For the average person, they are unnecessary.

Randomizing the serial is to use the same amiibo twice, such as with Breath of the Wild.

Foomiibo is not explained for a reason. Think about what it does and then consider why it would be left out of the guide.

AbandonedCart, 2021-12-19:
I'm not entirely sure how the project has managed to survive over 5 years without this ever being an issue in the past, but it appears the current generation of users are much more dependent on step-by-step guides than in the past. 

That being the case, I will add some general explanations of the individual properties to the guide site. It may take some time to be the comprehensive guide expected, since the other big change over the last 5 years is the lack of contributors.

closed 2021-12-19 by AbandonedCart