HiddenRamblings/TagMo #380: App loads, crashes on crypto key setup

Tiberon, 2021-12-20:
Me again, thanks for the quick commit @AbandonedCart 
Google Pixel 6 Pro,

Logcat below
--------------------------

Google Pixel 6 Pro
Android SDK 31 (12)
TagMo 92194be

--------- beginning of crash
12-20 14:34:15.825 13445 13445 E AndroidRuntime: FATAL EXCEPTION: main
12-20 14:34:15.825 13445 13445 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 13445
12-20 14:34:15.825 13445 13445 E AndroidRuntime: java.lang.IllegalStateException: Fragment SettingsFragment{90fb23} (aa05f9f0-013f-4960-9f76-db8d29e80cd9 id=0x7f09016a) not attached to a context.
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at androidx.fragment.app.Fragment.requireContext(Fragment.java:900)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at androidx.fragment.app.Fragment.getResources(Fragment.java:964)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at androidx.fragment.app.Fragment.getString(Fragment.java:986)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.onImportKeysClicked(SettingsFragment.java:371)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$locateKeyFiles$37$com-hiddenramblings-tagmo-BrowserActivity(BrowserActivity.java:1954)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda38.onClick(Unknown Source:2)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.google.android.material.snackbar.Snackbar$1.onClick(Snackbar.java:351)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.view.View.performClick(View.java:7441)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.view.View.performClickInternal(View.java:7418)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.view.View.access$3700(View.java:835)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.view.View$PerformClick.run(View.java:28676)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
12-20 14:34:15.825 13445 13445 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
--------- beginning of main
12-20 14:35:00.683 16027 16027 I ramblings.tagm: Late-enabling -Xcheck:jni
12-20 14:35:00.742 16027 16027 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10455; state: ENABLED
12-20 14:35:00.822 16027 16027 V GraphicsEnvironment: ANGLE Developer option for 'com.hiddenramblings.tagmo' set to: 'default'
12-20 14:35:00.822 16027 16027 V GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
12-20 14:35:00.822 16027 16027 D NetworkSecurityConfig: No Network Security Config specified, using platform default
12-20 14:35:00.823 16027 16027 D NetworkSecurityConfig: No Network Security Config specified, using platform default
12-20 14:35:00.856 16027 16046 D vulkan  : searching for layers in '/data/app/~~kVdNpDkLJUOHFsQ1HTy63w==/com.hiddenramblings.tagmo-Rl4aVxu0jbLjxKqSP2goiw==/lib/arm64'
12-20 14:35:00.856 16027 16046 D vulkan  : searching for layers in '/data/app/~~kVdNpDkLJUOHFsQ1HTy63w==/com.hiddenramblings.tagmo-Rl4aVxu0jbLjxKqSP2goiw==/base.apk!/lib/arm64-v8a'
12-20 14:35:00.860 16027 16046 D mali_cmarp_predictor: checking cmar_predictor for com.hiddenramblings.tagmo
12-20 14:35:00.912 16027 16027 W ramblings.tagm: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
12-20 14:35:00.913 16027 16027 W ramblings.tagm: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
12-20 14:35:00.973 16027 16027 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService()
12-20 14:35:00.974 16027 16027 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService() returned 0x0
12-20 14:35:00.974 16027 16027 D RenderScript HIDL Adaptation: Using Fallback Path.
12-20 14:35:00.980 16027 16027 D ramblings.tagm: Successfully queried cache dir: /data/user_de/0/com.hiddenramblings.tagmo/code_cache
12-20 14:35:00.980 16027 16027 D RenderScript: Setting cache dir: /data/user_de/0/com.hiddenramblings.tagmo/code_cache
12-20 14:35:02.576 16027 16027 D AndroidRuntime: Shutting down VM
12-20 14:35:02.577 16027 16027 E AndroidRuntime: FATAL EXCEPTION: main
12-20 14:35:02.577 16027 16027 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 16027
12-20 14:35:02.577 16027 16027 E AndroidRuntime: java.lang.IllegalStateException: Fragment SettingsFragment{6af8a25} (b785962c-0595-4f8d-b39e-4b43bb544505 id=0x7f09016a) not attached to a context.
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at androidx.fragment.app.Fragment.requireContext(Fragment.java:900)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at androidx.fragment.app.Fragment.getResources(Fragment.java:964)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at androidx.fragment.app.Fragment.getString(Fragment.java:986)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.onImportKeysClicked(SettingsFragment.java:371)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$locateKeyFiles$37$com-hiddenramblings-tagmo-BrowserActivity(BrowserActivity.java:1954)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda38.onClick(Unknown Source:2)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at com.google.android.material.snackbar.Snackbar$1.onClick(Snackbar.java:351)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.view.View.performClick(View.java:7441)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.view.View.performClickInternal(View.java:7418)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.view.View.access$3700(View.java:835)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.view.View$PerformClick.run(View.java:28676)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
12-20 14:35:02.577 16027 16027 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
[... 18 lines cut ...]
12-20 14:35:02.577 16027 16027 E UncaughtException: 	at android.os.Looper.loop(Looper.java:288)
12-20 14:35:02.577 16027 16027 E UncaughtException: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
12-20 14:35:02.577 16027 16027 E UncaughtException: 	at java.lang.reflect.Method.invoke(Native Method)
12-20 14:35:02.577 16027 16027 E UncaughtException: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
12-20 14:35:02.577 16027 16027 E UncaughtException: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
12-20 14:35:02.577 16027 16027 W System.err: java.lang.IllegalStateException: Fragment SettingsFragment{6af8a25} (b785962c-0595-4f8d-b39e-4b43bb544505 id=0x7f09016a) not attached to a context.
12-20 14:35:02.577 16027 16027 W System.err: 	at androidx.fragment.app.Fragment.requireContext(Fragment.java:900)
12-20 14:35:02.577 16027 16027 W System.err: 	at androidx.fragment.app.Fragment.getResources(Fragment.java:964)
12-20 14:35:02.577 16027 16027 W System.err: 	at androidx.fragment.app.Fragment.getString(Fragment.java:986)
12-20 14:35:02.577 16027 16027 W System.err: 	at com.hiddenramblings.tagmo.settings.SettingsFragment.onImportKeysClicked(SettingsFragment.java:371)
[... cut ...]
12-20 14:35:02.578 16027 16027 W System.err: 	at android.os.Looper.loop(Looper.java:288)
12-20 14:35:02.578 16027 16027 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
12-20 14:35:02.578 16027 16027 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
12-20 14:35:02.578 16027 16027 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
12-20 14:35:02.578 16027 16027 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

Tiberon, 2021-12-20:
Issue persisting after commit 09809f4

Logcat below
----------
Google Pixel 6 Pro
Android SDK 31 (12)
TagMo 09809f4

--------- beginning of crash
12-20 15:12:54.931 24189 24189 E AndroidRuntime: FATAL EXCEPTION: main
12-20 15:12:54.931 24189 24189 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 24189
12-20 15:12:54.931 24189 24189 E AndroidRuntime: java.lang.ClassCastException: com.google.android.material.snackbar.SnackbarContentLayout cannot be cast to com.google.android.material.snackbar.Snackbar
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$locateKeyFiles$37$com-hiddenramblings-tagmo-BrowserActivity(BrowserActivity.java:1954)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda38.onClick(Unknown Source:2)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at com.google.android.material.snackbar.Snackbar$1.onClick(Snackbar.java:351)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.view.View.performClick(View.java:7441)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.view.View.performClickInternal(View.java:7418)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.view.View.access$3700(View.java:835)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.view.View$PerformClick.run(View.java:28676)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:201)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:288)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
12-20 15:12:54.931 24189 24189 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
--------- beginning of main
12-20 15:27:41.895 29947 29947 I ramblings.tagm: Late-enabling -Xcheck:jni
12-20 15:27:41.975 29947 29947 D CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10456; state: ENABLED
12-20 15:27:42.066 29947 29947 V GraphicsEnvironment: ANGLE Developer option for 'com.hiddenramblings.tagmo' set to: 'default'
12-20 15:27:42.066 29947 29947 V GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
12-20 15:27:42.067 29947 29947 D NetworkSecurityConfig: No Network Security Config specified, using platform default
12-20 15:27:42.067 29947 29947 D NetworkSecurityConfig: No Network Security Config specified, using platform default
12-20 15:27:42.119 29947 29972 D vulkan  : searching for layers in '/data/app/~~8AG0ARhN_-m79gQDE0UMqA==/com.hiddenramblings.tagmo-x5t3BqpEWUUazDUr6wrS5A==/lib/arm64'
12-20 15:27:42.120 29947 29972 D vulkan  : searching for layers in '/data/app/~~8AG0ARhN_-m79gQDE0UMqA==/com.hiddenramblings.tagmo-x5t3BqpEWUUazDUr6wrS5A==/base.apk!/lib/arm64-v8a'
12-20 15:27:42.123 29947 29972 D mali_cmarp_predictor: checking cmar_predictor for com.hiddenramblings.tagmo
12-20 15:27:42.200 29947 29947 W ramblings.tagm: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
12-20 15:27:42.200 29947 29947 W ramblings.tagm: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
12-20 15:27:42.262 29947 29947 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService()
12-20 15:27:42.263 29947 29947 D RenderScript HIDL Adaptation: IRenderScriptDevice::getService() returned 0x0
12-20 15:27:42.263 29947 29947 D RenderScript HIDL Adaptation: Using Fallback Path.
12-20 15:27:42.270 29947 29947 D ramblings.tagm: Successfully queried cache dir: /data/user_de/0/com.hiddenramblings.tagmo/code_cache
12-20 15:27:42.270 29947 29947 D RenderScript: Setting cache dir: /data/user_de/0/com.hiddenramblings.tagmo/code_cache
12-20 15:27:43.993 29947 29960 W System  : A resource failed to call end. 
12-20 15:27:43.993 29947 29960 W System  : A resource failed to call end. 
12-20 15:27:44.167 29947 29947 D AndroidRuntime: Shutting down VM
12-20 15:27:44.168 29947 29947 E AndroidRuntime: FATAL EXCEPTION: main
12-20 15:27:44.168 29947 29947 E AndroidRuntime: Process: com.hiddenramblings.tagmo, PID: 29947
12-20 15:27:44.168 29947 29947 E AndroidRuntime: java.lang.ClassCastException: com.google.android.material.snackbar.SnackbarContentLayout cannot be cast to com.google.android.material.snackbar.Snackbar
12-20 15:27:44.168 29947 29947 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity.lambda$locateKeyFiles$37$com-hiddenramblings-tagmo-BrowserActivity(BrowserActivity.java:1954)
12-20 15:27:44.168 29947 29947 E AndroidRuntime: 	at com.hiddenramblings.tagmo.BrowserActivity$$ExternalSyntheticLambda38.onClick(Unknown Source:2)
12-20 15:27:44.168 29947 29947 E AndroidRuntime: 	at com.google.android.material.snackbar.Snackbar$1.onClick(Snackbar.java:351)
12-20 15:27:44.168 29947 29947 E AndroidRuntime: 	at android.view.View.performClick(View.java:7441)
[... cut ...]
12-20 15:27:44.169 29947 29947 W System.err: 	at android.os.Looper.loop(Looper.java:288)
12-20 15:27:44.169 29947 29947 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:7839)
12-20 15:27:44.169 29947 29947 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
12-20 15:27:44.169 29947 29947 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
12-20 15:27:44.169 29947 29947 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

AbandonedCart, 2021-12-21:
Thanks for the logs and donation. I believe the last commit actually solved the issue, but I opened the issue back up just to be sure. It was being closed automatically by the commits. It took a bit to actually respond, but I figured pushing through the fixes would be better than waiting until I got back.

commit 09809f4d (2021-12-20): Discard a relatively pointless automation (Fixes #380)
commit d06f0ab7 (2021-12-20): Set view visibility on the UI thread (Fixes #379)

Discard a relatively pointless automation (Fixes #380)

Fix the UI, despite the implication of it appearing

This UI should really never be reached, unless setup has not been completed properly and the keys don't exist. It also shouldn't crash to avoid disguising the real issue of no keys.
commit 34a04565 (2021-12-22): Set view visibility on the UI thread (Fixes #379)

Discard a relatively pointless automation (Fixes #380)

Fix the UI, despite the implication of it appearing

This UI should really never be reached, unless setup has not been completed properly and the keys don't exist. It also shouldn't crash to avoid disguising the real issue of no keys.

Urge issues to be posted on GitHub, not Amiibomb

Since the project is not hosted on Amiibomb, the moderators of Amiibomb are unresponsive, and the community usually has counterproductive advice, let's urge posting issues where they have a legitimate chance [... cut]
closed 2021-12-22 by AbandonedCart