HiddenRamblings/TagMo #387: Amiibo bin not showing up

ProfessorDragon, 2022-01-04:
I connected my computer to my phone using USB.
I scanned my amiibo and then copied the bin from Download/TagMo/Backups to my computer.
Next, I decrypted the bin and set the defense value to 0 using odwinc's SSBU amiibo editor.
Then I encrypted it, and put that bin in Download/TagMo/Backups.
I restarted the app, and the new bin file is not showing up, which means I cannot write the data back to my amiibo.
What am I doing wrong?
The bin files are here: [amiibo_bins.zip](https://github.com/HiddenRamblings/TagMo/files/7805828/amiibo_bins.zip)

AbandonedCart, 2022-01-04:
None of what you are doing seems to be in TagMo and there are notices everywhere about not sharing or requesting bin files, let alone posting them directly in an issue.

ProfessorDragon, 2022-01-04:
Sorry, I thought I was meant to share the bins so you guys could see if there was anything wrong with them.
The editor seems to be working fine, as the editor is able to decrypt the new encrypted bin without any problems.

AbandonedCart, 2022-01-04:
I'm a bit skeptical with all the warnings on that project about how it doesn't work with TagMo.

AbandonedCart, 2022-01-04:
Have you tried using the the capture logcat option after they wouldn't load? That should give an explanation.

ProfessorDragon, 2022-01-04:
Yes, I tried using capture logcat, and it doesn't show anything related to the issue. One time it said 'data does not start with 0x04', which is not true, as I checked and both of the bins do start with 0x04.
Regarding the 'note on tagmo' in the editor - I had installed TagMo 047bd2e and all of the amiibos showed up fine. I think it may be the latest update that is causing the problem, and I cannot find a way to redownload the previous version.

AbandonedCart, 2022-01-04:
I'm a bit confused by how the latest update could have caused the issue. Since the quoted hash, the commits have all been in the NFC activity, most of which ended up actually being reversed along the way.

ProfessorDragon, 2022-01-04:
I don't know either why the update would cause the issue, but that's literally the only thing which has changed since it was working with the previous update.

AbandonedCart, 2022-01-04:
Well, I guess I will just keep pushing forward until I can find some evidence that changing a Dialog to a Snackbar causes some sort of bug in the Android API. Hopefully it sorts itself out along the way.

ProfessorDragon, 2022-01-04:
Could you add an option to disable the built-in amiibo explorer in the next version? Like 2.7.0.

AbandonedCart, 2022-01-04:
It's not a file explorer anymore. That has been completely removed.

AbandonedCart, 2022-01-05:
With all the recent changes and the timing of the issue, it may be good to force a reload of your keys. That can be done by manually selecting the import option in the app settings. Beyond that, you may also want to make sure the files still have a ".bin" extension. That is the only requirement for the app to attempt loading them.

ProfessorDragon, 2022-01-06:
What I was saying is that you should add the file explorer back in. I was able to load the new bin file and write it to a tag successfully in 2.7.0 with the file explorer.

AbandonedCart, 2022-01-06:
Even if it weren't a massive amount of code, it would have the same end result. Nearly every tutorial, video, post, etc about the older TagMo versions starts off "disable the file explorer"

ProfessorDragon, 2022-01-06:
Alright, but I still think it would be useful to have the option.

AbandonedCart, 2022-01-06:
I can see where saying "it won't change the result" may seem like laziness. It's not. 

The only difference between a file browser and the automated scan is what is shown to the end user and who selects the file. In a file browser, you see all your files and pick the one that you want. It still has to be a bin and still has to pass the checks. In the scan, all the files are gathered up and only those ending in "bin" go through the check. All that pass are shown. 

Adding a file browser to "solve" this issue will be fundamentally dead weight. The easier route would be to add intent filters that allow an external file browser to open files with TagMo. It's the same core functionality without the wasted effort reinventing the wheel, or in this case, the file browser. Whether or not it will actually change anything is a very slim margin of possibility.

closed 2022-01-10 by AbandonedCart