HiddenRamblings/TagMo #388: PL_pl - translation to Polish

NoiRioN, 2022-01-04:
Hi,

I really appreciate your work, so I decided to translate the strings right after I saw your call for it. Please find it attached. Hopefully it's useful for the project.

Please remember to take a good care of latin-ext characters coding in Polish: ĘęÓóĄąŚśŁłŻżŹźĆćŃń - only this way you'll avoid blank or unknown characters in the app, and this happens a lot on Android. I saw the xml is in UTF-8, therefore I believe it should be okay, but still please check it twice.

Best regards,
Chris
a.k.a.: NoiRioN, NoiRioN_7, NoiRioN.7, NRN7
Socials: [IG](https://www.instagram.com/noirion.7/) & [Twitch](https://www.twitch.tv/noirion_7)

[strings.zip](https://github.com/HiddenRamblings/TagMo/files/7809591/strings.zip)

AbandonedCart, 2022-01-04:
It doesn't appear to have popped up with any invalid characters, but let me know if anything doesn't show up and I can investigate further.

closed 2022-01-04 by AbandonedCart

AbandonedCart, 2022-01-04:
Oh, and thanks! Sorry, got a little wrapped up and forgot to be a human.

NoiRioN, 2022-01-04:
Remember to put translators in credits and you're forgiven 😁