HiddenRamblings/TagMo #389: Crash when selecting "delete amiibo"

riley-west, 2022-01-06:
Steps to reproduce:

Scan amiibo
Press menu "three dots" button
Select "delete amiibo"

App hangs for approx 30 seconds then crashes out.

Log attached. 
[crash_logcat.txt](https://github.com/HiddenRamblings/TagMo/files/7819333/crash_logcat.txt)

AbandonedCart, 2022-01-06:
You can't delete something that doesn't exist yet. I'm pretty sure you knew that going into it, but congrats on finding something to report.

closed 2022-01-06 by AbandonedCart

riley-west, 2022-01-06:
I mostly wondered what it was going to do. Given that it had not opened a bin file and was a physical amiibo. Would it:

Do nothing.
Reset the amiibo back to defaults.

I most certainly did not know that going into it.